Ralim/IronOS #1663: disable the opening of the constant temperature control with the (+/A) button.

Dekanser, 2023-04-11:
Problem.
When I solder, I often have to use boost mode. The temperature limit for boost mode is set to 340 degrees, but I don't always need this temperature. More often I use boost mode for 0.5-2 seconds to raise the temperature by 40-60 degrees.

Very often, when using boost mode for a short time, instead of boost mode, the temperature control interface opens.
[image]

Without visual control, I cannot understand whether boost mode or temperature control has turned on, since both actions are performed with one button (+ / A).

As a result, I constantly unintentionally set the maximum possible temperature to 400 degrees instead of briefly using boost mode.

---
What do we need.
I would like to have a checkbox to disable the opening of the constant temperature control with the (+/A) button.
At the same time, it will remain possible to set the desired constant temperature by opening the temperature adjustment interface with the button (-/B)
The (+/A) button in soldering mode will only boost the mode, and if the user needs to change the constant temperature, the user presses the (-/B) button and then uses (+/A) to increase or (-/B) to decrease the temperature .

---
Alternatives.
I tried using the button lock
"Setting: Allow locking buttons"
but this is not at all what I need, since it blocks not only the (+ / A) button, but also (- / B). I lose the ability to set the desired constant temperature by opening the temperature adjustment interface.

---
Why is it important?
I believe this is an essential feature that will help protect the user from erroneously setting the maximum temperature to 400 degrees, which can lead to damage to the soldered device or element, as well as to the heating element.
Without visual control of the screen, there is no way to determine what you are doing, briefly turn on the boost mode or change the constant temperature by 400 degrees, which causes a lot of problems.
The user should be able to separate this functionality more explicitly, since the delay in pressing does not provide reliable control.
Ideally, this setting should be enabled by default to help protect the user from making mistakes.
Reaching for the button (- / B) does not cause such inconvenience as 400 degrees turned on by accident.

discip, 2023-09-23:
@Dekanser
Since your request is not that hard to achieve, but at the same time relatively special, which in turn makes it not convenient to implement as rightly stated by @ia in https://github.com/Ralim/IronOS/issues/1719#issuecomment-1615208169, I recommend you to create a fork and change the corresponding line in code yourself:

https://github.com/Ralim/IronOS/blob/b489601883adc1bf241526eef91e46077bf18a6f/source/Core/Threads/OperatingModes/Soldering.cpp#L82-L83

You just need to enter `break;` between line 82 and 83.
That's it. 😊

nonokirby, 2025-01-20:
@discip where would this line be located in the current version? I'm looking to do the same, but the file has been moved since the version mentioned above.

nonokirby, 2025-01-21:
https://github.com/nonokirby/IronOS/releases/tag/V2.23-RC1-Custom I created a release in my fork for 2.23-RC1 for those who see this later on to reference. i will continue to update my fork alongside the official release as best as i can.

ia, 2025-01-23:
Hello, @nonokirby .

> Ideally, this setting should be enabled by default to help protect the user from making mistakes.

Please, **no**. And it's not even about this project only. When there are a lot of users already and they did get use to the _current_  buttons' logic, you just can't break the defaults for the buttons.

I'm not even talking about how many bug reports we will start to get if it will be rolled out with the new defaults that now _"I can't increase the soldering temperature in the soldering mode"_, since it's not duty of people to read every documentation update and/or release notes.

> Alternatives.

As of for the alternatives. There is the option already for locking mode to block A/B buttons but not boost mode. Have you ever tried this option as the workaround for your problem? If yes, then what was wrong exactly with this alternative (since adding more and more new code may getting "costly" from the point of flash storage for some supported hardware)? Just would like to know and to get some feedback on that as well. Thanks.