Ralim/IronOS #1668: Pinecil v2 reboots on PD after enable soldering

no111u3, 2023-04-21:
**Describe the bug**
Pinecil v2 plugged by standard orange USB-C cable to Samsung PD charger EP-T6530 in socket with PD reboot since temperature higher than 100 °C. Try on original Apple PD charger — works good.

**To Reproduce**

**Expected behavior**

Works as another PD charger.

**Details of your device:**
 
 - Device: Pinecil v2
 - Release: 2.21
 - Power adapter being used: Samsung EP-T6530

**Additional context**
Change cable to another has no effect.

StevenSeifried, 2023-04-21:
I own the same charger and can confirm the issue with the following USB-PD cables
- Orico 240W cable (2m)
- Toocki 240W cable (2m)
- Toocki 60W cable (1m)

Unfortunately, the charger is a bit picky regarding cables. The only cable with which I haven't had any problems so far with multiple devices is the Samsung DX510

Ralim, 2023-07-23:
Can you re-test the latest dev builds when you get a chance, this may have been resolved by some PD tweaks done a little while ago 🙇🏼

discip, 2023-09-23:
Closing since no one responds.

S-D-, 2024-05-12:
I have the same charger. This issue still exists with the latest release version and with latest firmware built from the dev branch.
I don't have this issue with PinePower charger.

Ralim, 2024-06-02:
Can you test this build: https://github.com/Ralim/IronOS/actions/runs/9328320633

It has #1917 included, which allows you to change the PD mode to different selections. This should help you narrow down the issue for your specific power adaptor (and work around it)

no111u3, 2024-06-02:
@Ralim sorry for long delay, I tested with firmware from your last link and have same problem.

no111u3, 2024-06-02:
This issue only for slot with 19 V for slot with 10 V max solder works perfectly.

Ralim, 2024-06-02:
Did you adjust the PD mode setting in the power menu?
The Safe profile should improve compatibility.

no111u3, 2024-06-02:
The same behaviour on the safe PD mode too.

no111u3, 2024-06-02:
But interesting part: when I select no dynamic works better than before.

Ralim, 2024-06-02:
@no111u3 

Can you please provide:
1. Image of ratings on side of the charger
2. List of the profiles decided in the USB-PD debug menu (hold front button when connecting Pinecil)
3. What voltage is negotiated after changing the PD mode setting and power cycling for each menu option

To try and diagnose what the next steps are

discip, 2024-07-24:
@no111u3
@S-D-
Could please one of you provide the according data?

nisembedded, 2024-07-25:
@Ralim @S-D- Answer instead @no111u3 

1.
[image]
2.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
3.
Usual 19.9-20 but stable work with I selected mode with no dynamics selected voltage is 15 V.

Ralim, 2024-07-28:
So the only issue I can see is the PPS profile being decoded as offering 5-16V 3A when the case shows 5-16V 2.8A.
All others decode fine os may be an issue of the case not matching whats programmed inside.

I'm assuming your using port C1 (higher powered port).

Given that its decoding this fine, this may just be a charger that is hyper picky / overly unhappy with the power draw profile of the Pinecil and so using no-dynamics forces different profiles which work.