Ralim/IronOS #1670: MHP30 can shutdown and not enter soldering mode until it is moved

codingcatgirl, 2023-04-23:
When the MHP30 has not been moved for a while, it will enter shutdown mode. Once it is in this mode, attempting to enter soldering mode will immediately bring you back to the Main Menu.

The issue can be circumvented by picking up the MHP30 and moving it around to recover from the shutdown mode – or by disabling shutdown entirely.

I would have fixed this issue myself, but i am not sure why shutdown mode is a thing at all on this device, as you would expect that it is not moved for a long time while in use.

Ralim, 2023-04-23:
Feels like a bug; the intent is that it shuts down _on movement_ (so if you knock it over it should shutdown).

codingcatgirl, 2023-04-23:
Right, it does that too though, so there's that :)

discip, 2023-06-29:
@codingcatgirl
What about this one?
Is this solved now?
If it is, please close it. 😊

codingcatgirl, 2023-06-30:
Oof, no idea. My contribution to the firmware was long enough ago that i don't remember if i tried to fix it. My comment in #1570 seems to imply that i have not fixed this part of the issue however.

discip, 2023-06-30:
> My comment in #1570 seems to imply that i have not fixed this part of the issue however.

Are you planning to solve this?

codingcatgirl, 2023-06-30:
No, i only fixed the other part of the issue as a side effect of the new
feature i contributed.

Ralim, 2023-07-23:
@codingcatgirl Can you confirm that your MHP30 has the MSA301 accelerometer in it?
Going to try and reproduce this shortly :)

codingcatgirl, 2023-07-24:
Can't confirm this right now, but i am pretty sure that looking at the code this didn't look like a hardware-related issue but simply due to the fact that if there hasn't been movement in a while it will automatically exit sodlering mode again right after entering it.

PR #1762 (2023-08-04): Use 3 count filter for MHP30 acceleromter

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**

Requires it to trip 3 times in a row to fire. So really only knocking the unit over trips it off.
Also moves shutdown timer forwards when we shutdown timeout, and defaults shutdown timeout to being off (rather than 10 mins)
Should fix #1670 and also prevent nuisance trips.

@codingcatgirl  Would be amazing if you could give this a spin when you get time 🙇🏼
closed 2023-08-04 by Ralim