Ralim/IronOS #1671: MHP30 adressable LED only shows green channel

codingcatgirl, 2023-04-23:
**Describe the bug**
The Code suggests that the MHP30 shouuld show different LED colors during operation. However, the only colors i see while using it are green, dark green (when it should be orange) and off (when it should be red). It looks like adressing the LED doesn't work.

Ralim, 2023-04-23:
For both of my devices here it is showing R/G/B as expected on the latest release :/
Can you confirm that your led is fully operational via the factory firmware?
Also any chance of a photo of the led (wondering if they have changed it)

codingcatgirl, 2023-04-23:
What i can say is that it definitely lights up in blue in the DFU mode. Not sure if i've seen red, i didn't pay *that* close attention to it since i got it 2 days ago :D Are there any instructions how to best open up this thing? then i'll take a look

Ralim, 2023-04-23:
The led is is under the hot top so can just pull up on the heater pad and it should neatly pop off the base (its on pins).

Okay so definitely sounds like something is going wrong somewhere hmm.

codingcatgirl, 2023-04-23:
Thanks, that was almost too easy.

Anyway, the LED does look like then one that has been indentified before though
[image]

codingcatgirl, 2023-04-24:
I've set it to white and it is cyan now. Hm.

Ralim, 2023-04-24:
Definitely a new revision of that top board compared to mine.
Gotta wonder what changed lol.

Maybe try adjusting the timing values a bit in the driver code? Could be on the edge for the bit timings for your unit?

codingcatgirl, 2023-04-24:
I've flashed the original firmware again. It seems that the original firmware can't show any red either. So i have to assume that my unit is either broken… or that the new hardware revision does have some difference and the firmware they have available for download isn't correct for my hardware revision. Wish i would have paid more attention to my first test with the stock firmware. The latter would be better of course, but it might just be time to see if i can swap the LED, unless anyone else will show up with the same problem here.

Ralim, 2023-04-25:
Damn, I was really hoping it wasnt a damaged led.
On the bright side its not stupid hard to replace at a guess.

In terms of timings; its all the asm crud in https://github.com/Ralim/IronOS/blob/dev/source/Core/Drivers/WS2812.h

codingcatgirl, 2023-04-25:
> On the bright side its not stupid hard to replace at a guess.

Yeah especially since i now have a hot plate… ……… oh

discip, 2023-06-29:
@codingcatgirl
Did you manage to solve this issue?
If so, please don't forget to close this. 😊

codingcatgirl, 2023-06-30:
The issue seems to be hardware-related, so i'm gonna glose this, yep.

closed 2023-06-30 by codingcatgirl

codingcatgirl, 2023-10-26:
Do you see any other colors with the official firmware? If not, you just have the same broken LED as me and this is not a firmware issue.