Ralim/IronOS #1673: Power limit compatibility with PD-adapter on TS100

ArtRoman, 2023-04-24:
**Describe the bug**

Power limit polling(?) not compatible with PD-adapter on TS100. Any value (even 5W) makes PD-meter show 0.0W on powerful bricks, and makes less powerful bricks to turn off when start heating.
I'm using Type-C (PD) to barrel jack adapter (19V, 60W I believe).

Kuulaa 20W (12V max) is working, slow heating
Kuulaa 33W (20V max) and Apple MBA charger 30W (20V max) are not working, turns off
Kuulaa 65W (20V max) is working, fast heating
ZMI QB818 power bank 30W is working (15V max), medium heating speed
Energizer xp20002pd power bank 65W is working (20V max), fast heating

**To Reproduce**
1. Go to Advanced settings, set Power limit to 10W
2. Return to home screen, click on button closest to the tip to start heating
3. Some power bricks turns off, I think because of power consumption protection. Some are working, but cable with power meter display shows "0.0W PD"
4. Without power limit it consumes 28W from the 30W power bank, which is OK.

**Expected behavior**
Soldering tip should start heating slowly without overcurrent protection, according to selection power settings

**Details of your device:**
 
 - Device: TS100
 - Release: 2.21.5D96470
 - Power adapter being used: a set of, described earlier
 - If this is an accelerometer related issue, please include its model number here: no

Settings: 
[image]

Heating up with 5W limit: 
[image]

Heating with no power limit:
[image]

Ralim, 2023-04-25:
Hi, the power limit is a average power limit. To be used for keeping the average power below a limit to prevent overheating of the source. It can't limit the peak current which is what trips usb-pd power adaptors.

pklapperich, 2023-04-25:
@Ralim Between your comment and the [wiki](https://github.com/Ralim/IronOS/wiki#differences-at-a-glance), it sounds like this is a limitation of IronOS in general and probably not something that can be solved in software, is that correct?

I ask because help text in the settings section says ["Maximum"](https://github.com/Ralim/IronOS/blob/da18b9b60f6b4f4f9b23a03f6c5366e79d1a978b/Translations/translation_EN.json#L220), not "Average". Perhaps this should be viewed as a documentat/translation bug. Maybe a change in the help text to "Average Maximum power the iron can use (W=watt)" would be good.

Ralim, 2023-04-25:
It's not a limitation of IronOS, but rather a limitation of the Iron Hardware. The MHP30 is the only device with proper max power limiting.

But yeah this sounds like a documentation issue again.

PR #1727 (2023-06-29): correct wording for 'max power'
For details see #1673.

closes #1673
closed 2023-06-29 by discip