Ralim/IronOS #1683: Bluetooth unable to pair

kiesstein, 2023-05-14:
I have got the PinecilV2 and flashed the firmware v2.21 to try out BLE.
The device shows up on 2 phones and my pc as "Pinecil-2210595" but when i try to pair, it immediately cancels the pairing process:
- For Android when i tab on pair it shows for a fraction of a section "Pairing..." and that wast it.
- For KDE Bluetooth-Wizard says "The setup of Pinecil-2210595 has failed."

How can i know if it is a HW problem or SW? What can i do to debug this issue further?
I appreciate any tips and help.

Ralim, 2023-05-14:
The device is BLE only, so no real pairing to do here.
A BLE aware app should be able to just connect to it.

kiesstein, 2023-05-14:
omg thank you, i just tried with the chrome website and it works.
TY

closed 2023-05-14 by kiesstein