Ralim/IronOS #1688: temperature fluctuations pencil v2

elpidifor, 2023-05-22:
my setup: 
  original pinecil v2 (tried 3 different)
  firmware (3 different): 2.20 stock , 2.20 release and 2.21 release 
  original pine64 original short cartridge (tried 3 different)
  cable tried 3: original pine64 1.5m type-c and baseus 2m 100W type-c (tried 2 different)
  power supply tried 2: voltme 140w and meanwell lrs-350-24 (24v 350W)
  
after all the advice in pine64 support and pinecil chat temperature still range 340 to 360 when 350 was set

I clean all contact with IPA
did settings reset
calibrate temperature after 30 minutes of being turned off
heated to 350 stay for a few minutes then cool and again heated and repeated several times

also Tip K is unstable in the cold state and fluctuates +- 100 and in the heated state also has large fluctuations

[image]

Ralim, 2023-06-02:
The raw tip reading will always have a few 10's to 100's of microvolts of instability due to how this all works.

The slow temperature oscillation is "normal" at the moment as the V2 needs a re-tune after a bunch of work that does done earlier. Just haven't had time to setup my unit to adjust the tuning values for the stabilisation filter.

Ralim, 2023-06-04:
Have done another tune, please try flashing the latest dev build and let me know how it performs for you

cohaolain, 2023-06-05:
Just tried e7bcf92 here on my Pinecil v2, with a target temperature of 360C I'm getting fluctuations with range between 350C and 380C, or wider. The period between power bursts seems to be quite long, roughly 3s or so - is that expected to still be so wide/long? Still seems to be much less smooth than before #1485 / #1535 (aside from the fluctuations solved by that PR, of course).

[image]

Thanks for your attention on this :)

Ralim, 2023-06-05:
Hia,

Can you please confirm:
Power pulses are turned off in the menu
What tip your using
What your ambient temperature is
And
What power supply your using ?

cohaolain, 2023-06-10:
> Hia,
> 
> Can you please confirm: Power pulses are turned off in the menu What tip your using What your ambient temperature is And What power supply your using ?

Just tested this again here to ensure everything lined up:
- Ambient 22.6C
- Target 360C
- Observed temperature dropping to 345C and then jumping to 380C, with a period of (roughly) ~3s
- Tip during this test is an ST-KU, cleaned with IPA.
- Power pulses disabled
- Power supply is a PinePower GaN (current revision), 65W available @ 20V

Ralim, 2023-06-12:
Hmm, okay thats odd as it is not fluctuating like that for me, maybe a few (2 ish) C on the screen and tip tester shows smaller fluctuations again.

I dont have a ST-KU, just the longer ones or the ST-B2 tips here. One all of these it behaves well.

cohaolain, 2023-06-12:
```V2.21.E7BCF92
04-06-23```

cohaolain, 2023-06-12:
That's quite interesting indeed, I will try with an ST-B2 here but I can't imagine the difference will be so stark. Very curious. 

@elpidifor have you tested the new version and seen an improvement?

sddev0, 2023-08-14:
@Ralim: Can you reopen this issue? Which additional steps would be helpful to mitigate the issue?

Ralim, 2023-08-14:
>Which additional steps would be helpful to mitigate the issue?

Test new tuning values for the tip if you have time to iterate.

https://github.com/Ralim/IronOS/blob/20e085487e08b901c7f0923c0a98200535beca80/source/Core/BSP/Pinecilv2/BSP.cpp#L167
and
https://github.com/Ralim/IronOS/blob/20e085487e08b901c7f0923c0a98200535beca80/source/Core/BSP/Pinecilv2/BSP.cpp#L173

I have had a few pokes and can replicate the issue but have not yet done any re-tuning yet.

discip, 2023-08-14:
@elpidifor, @cohaolain, @sddev0
Please check the latest adaptations. 😊

MabaKalox, 2023-09-20:
Same behaviour as @sddev0 have
v2.22
Target temp - 295
Temp fluctuates between - 280-320

V2.21
Target temp - 295
Temp fluctuates between 293-297

Ralim, 2023-09-20:
Have you tested the PR looking to (reduce these)[https://github.com/Ralim/IronOS/pull/1814]?

dogtopus, 2023-09-22:
Just tried RC2. Didn't see any noticeable improvement compare to RC1. Running on PD power supply still gives around 7 degree swing and on EPR 10-12 degree.

Ralim, 2023-09-22:
Ah good to know the tip.
Could you confirm the following:
Which tip code is on the tip?
If it was made by Pine64, Miniware or Others?
What device your using for measuring the tip temperature ?

dogtopus, 2023-09-22:
It's a (probably genuine) Miniware TS-D24 I bought from Banggood ages ago.

I just used the internal temperature reading. Good tip thermometers are quite expensive.

Ralim, 2023-09-22:
Thanks for the check.
Just wanted to narrow it down, as I have a handful of tips from aliexpress here that dont have the miniware logo on them that have quite poor sintering inside which oscillate a lot worse than this and wanted to rule it out.

I do intend to try and tune this more, but just genuinely havent had the days to spend on it.

discip, 2023-09-22:
Have tested the latest https://github.com/Ralim/IronOS/commit/c0a5e244b93eba570c08bc73ecd2fe7cfc067636 with the stock short tip.

Here is the result:
[image]

discip, 2023-09-23:
https://github.com/Ralim/IronOS/commit/07893daec76af2e7c66d87af7bfa0441664371c3

[image]
[image]

discip, 2023-09-23:
Results with https://github.com/Ralim/IronOS/commit/07893daec76af2e7c66d87af7bfa0441664371c3:

[image]
[image]

dogtopus, 2023-09-23:
I think #1820 makes the temperature swing slightly less (overall I see 2-3 degC reduction) but -5~+10 degC swing on EPR is still quite large.

Maybe there's still something to be done on the PID so it improves things further especially on the EPR end but with PD +-5 degC swing makes it somewhat passable already.

dogtopus, 2023-10-02:
Just tested with all of my other tips. Some heavy tips bounce pretty wildly (specifically TS-K and the CNCKitchen TS100 heating element with large diameter threaded insert tip) but after that initial period Pinecil v2 is able to keep all of them at +-1degC. So looks like something is indeed wrong with my TS-D24. Maybe the thermocopule in it isn't binding with the tip properly or something. Not sure why it works normally on Pinecil v1 and TS100 but there is a pattern.

goga1978, 2023-10-12:
Всем привет.
1. Необходимо вводить список жал BC2, K, KU и других.
2. Сделать нормальную калибровку жала по температурным точкам (450, 350, 250, 150)
3. Ввести время обновления экрана 0.5сек , 1 сек , 2 сек. Чтобы показания не скакали.
4. Ввести фильтрацию, скажем берем 20 значений ,  сортируем по возрастанию и берем значение из середины
(246, 246 , 247, 247, 248, 248 , 248 , 249 , 249, 250, 250) = 248.

Hello everyone.
1. It is necessary to enter a list of BC2, K, KU and other stings.
2. Make a normal calibration of the tip by temperature points (450, 350, 250, 150)
3. Enter the screen refresh time 0.5sec, 1 sec, 2 sec. So that the readings do not jump.
4. Enter filtering, say we take 20 values, sort in ascending order and take the value from the middle
(246, 246 , 247, 247, 248, 248 , 248 , 249 , 249, 250, 250) = 248.

MayorBug, 2023-10-28:
With the 2.22 RC3 release I am getting random +100C spikes while heating up. Can upload few videos if requested.

gamelaster, 2023-10-28:
@MayorBug video would be very useful, thanks!

MayorBug, 2023-10-28:
Two clips here: https://drive.google.com/drive/folders/1KDNVOfujuvDA0yDg31WEw0SdJ5hT_KiB

You can see the +100C spikes around 220C

MayorBug, 2025-09-01:
I'd say 2.23 fixed this

sddev0, 2025-09-01:
I can confirm the temperature is absolutely stable with 2.23!

This is how it was before:
[image]

This is now with 2.23 :)
[image]

Thank you!!

Ralim, 2025-09-02:
Thank you for the feedback and confirmation!

commit e7bcf920 (2023-06-04): Refactor PinecilV2 Tuning

Closes #1688
closed 2025-09-02 by Ralim