Ralim/IronOS #1694: i want to initialize SC7A20 Through i2c, I want to calculate Accelerometer Accleration through it? I am stuck in initialize Please Check my Code and Help me out Iam Using Stm32f103c8t6 Microcontroller

afzalabdullah, 2023-06-02:
#define SC7A20_I2C_ADDRESS 0x1C
#define SC7A20_CTRL_REG1   0x20
#define SC7A20_OUT_X_H_REG 0x28
#define SC7A20_OUT_X_L_REG 0x29
#define SC7A20_OUT_Y_H_REG 0x2A
#define SC7A20_OUT_Y_L_REG 0x2B
#define SC7A20_OUT_Z_H_REG 0x2C
#define SC7A20_OUT_Z_L_REG 0x2D
#define WHO_AM_I 0x00
#define POWER_MODE 0x01
#define RANGE 0x02
#define DATA_RATE 0x03

void SC7A20_Init(void) {
  uint8_t  sendData;
  uint8_t check_val;
  // Read the WHO_AM_I register to check if the device is present.
  HAL_I2C_Mem_Read(&hi2c1, SC7A20_ADDRESS, WHO_AM_I, 1, &check_val, 1, 1000);

  // If the device is present, initialize it.
  if (check_val == 0x01) {
    // Set the power mode to standby.
    sendData = 0x00;
    HAL_I2C_Mem_Write(&hi2c1, SC7A20_ADDRESS, POWER_MODE, 1, &sendData, 1, 1000);

    // Set the measurement range to 16-bit.
    sendData = 0x00;
    HAL_I2C_Mem_Write(&hi2c1, SC7A20_ADDRESS, RANGE, 1, &sendData, 1, 1000);

    // Set the data rate to 100Hz.
    sendData = 0x00;
    HAL_I2C_Mem_Write(&hi2c1, SC7A20_ADDRESS, DATA_RATE, 1, &sendData, 1, 1000);
  }
}
int main()
{
SC7A20_Init();
while(1){

                  uint8_t x_axis_h, x_axis_l, y_axis_h, y_axis_l, z_axis_h, z_axis_l;
                  HAL_I2C_Mem_Read(&hi2c1, SC7A20_I2C_ADDRESS, SC7A20_OUT_X_H_REG, 1, &x_axis_h, 1, 100);
	  	  HAL_I2C_Mem_Read(&hi2c1, SC7A20_I2C_ADDRESS, SC7A20_OUT_X_L_REG, 1, &x_axis_l, 1, 100);
	  	  HAL_I2C_Mem_Read(&hi2c1, SC7A20_I2C_ADDRESS, SC7A20_OUT_Y_H_REG, 1, &y_axis_h, 1, 100);
	  	  HAL_I2C_Mem_Read(&hi2c1, SC7A20_I2C_ADDRESS, SC7A20_OUT_Y_L_REG, 1, &y_axis_l, 1, 100);
	  	  HAL_I2C_Mem_Read(&hi2c1, SC7A20_I2C_ADDRESS, SC7A20_OUT_Z_H_REG, 1, &z_axis_h, 1, 100);
	  	  HAL_I2C_Mem_Read(&hi2c1, SC7A20_I2C_ADDRESS, SC7A20_OUT_Z_L_REG, 1, &z_axis_l, 1, 100);

	  	 double x_axis,y_axis,z_axis;

	  //	   Calculate the acceleration values
	  	   x_axis = (x_axis_h << 4) | x_axis_l;
	  	   y_axis = (y_axis_h << 4) | y_axis_l;
	  	   z_axis = (z_axis_h << 4) | z_axis_l;
}
}
this is the code of Stm32cube ide 
I2C1_SCL(pb6) 
I2C1_SDA(pb7)

Ralim, 2023-06-03:
Hello,
This does not look like an IronOS related issue?
Additionally, please use the Issue template, and explain the relationship to IronOS?

Ralim, 2023-06-23:
Closing due to no answer.

closed 2023-06-23 by Ralim