Ralim/IronOS #1719: Hold "+" to turn on iron rather than click

BreakMate, 2023-06-23:
To turn on the soldering iron, a quick click of "+" activates it and starts heating it up. On the other hand, to turn off the iron, one must hold "-" briefly. I have found myself by mistake popping the temp change menu when trying to turn the device off and vice versa with the same menu when I want to turn it on. 

It would be nice if the default way to turn on and off the device was holding "+" or "-", maybe it could be a setting? Even if you can point me in the right direction in the source code so that I can try to compile my own firmware where this is the default.

Thank you for your time.

Ralim, 2023-06-25:
Code wise its here: https://github.com/Ralim/IronOS/blob/0a3dfb34ece5f7cea942bcf8b555b18ff48456d1/source/Core/Threads/OperatingModes/HomeScreen.cpp#L51

You would sale the long and short switchcase for the front button.

Not sure if any others have any interest in this, if there is interest we could add a setting.

BreakMate, 2023-06-25:
Thank you for pointing me in the right direction! Now my only problem is I'm struggling to compile the firmware. 

I've installed STM32, Docker, and WSL2 as it says in the development instructions. But I cannot make heads or tails of it.

Is there any way I could buy you coffee and you could compile the firmware?

discip, 2023-06-29:
@BreakMate
You do not need to compile this locally.
You could just fork this repo, activate actions and change the corresponding part of code.
After that you can download the compiled zip file which you will find in actions.

BreakMate, 2023-06-30:
@discip Thank you for your reply
Where do I find compiled zip in actions. Im sorry, Im very new and looking up tutorials about github actions. I was successful in forking the repo and changing the code

discip, 2023-06-30:
@BreakMate
1. Did you manage to activate **`Actions`** for your fork?
If not please do so, by clicking on the **`Actions`** tab in the upper part of your fork. If opened you will see an option to activate **`Actions`**.

2. You should have done this prior to changing the code. If you did not you need to delete the fork and create a new one. Then activate **`Actions`** and only after that change the corresponding code.

3. In **`Actions`** you will find the **`workflows`**, if you click the topmost name to the right of the green dot with a hook you will be presented with 11 **`Artifacts`** (zipped builds) on the bottom of the page.
Choose the one you need by clicking it.

Done!

ia, 2023-06-30:
> Docker, and WSL2 as it says in the development instructions. But I cannot make heads or tails of it.

Hello. Since you have WSL2 locally already, here is one of possible short instructions to make a local build which should work:
- configure & start some latest fresh stable release of some gnulinux distro inside WSL and get access to shell
- install git & docker inside gnulinux using command something like `apk / dnf install docker git`
- clone repo to some dir inside gnulinux environment using `git clone https://github.com/...`
- go to cloned top-level directory of IronOS and build firmware for all supported models:
```cd IronOS; docker compose run --rm builder /bin/bash /build/ci/buildAll.sh```
- produced binaries with firmware will be placed to `ci/artefacts` after compilation

Good luck.

ia, 2023-06-30:
P.S. Regardless this particular report.
🤔
Just a rant since we're here. I noticed that from time to time there are [feature requests](https://github.com/Ralim/IronOS/issues/1663) like "I would like to make X using button Y while in state/operating mode Z". And this comes from objective limitations - there are two buttons only to control everything. In fact, I think myself about rearranging some actions/options for buttons from time to time. And it's really awesome that IronOS _allows_ to make such customization _in theory_ and _on practice for your own builds_. However, it's pretty obvious that if every request will be turning into additional feature with separate setting in menu, then this will complicate the process of managing & supporting different combinations/modes/key presses... and setting menu could turn into endless scrolling of menus like "what you would like to do with this button in that mode". Just thinking out loud, don't have adequate solution myself unfortunately. Ideally, it could be just awesome for advanced users to have easy reliable way to configure an iron's buttons just like we can bind in editors like vim/emacs any action to any combination of keys even if it will be one compile-time configuration in case with soldering irons (just like with some/old software when you had to compile-in settings right into binary).

discip, 2023-09-23:
Seems to work for @BreakMate since he does not respond. 😊

closed 2023-09-23 by discip