Ralim/IronOS #1720: How to access the PD debug menu on TS80P

He1my, 2023-06-27:
is there a way to access the PD debug menu on a TS80P, I tried pressing the button closest to the USB port and plugging in the cable, but that only goes to the regular debug menu to the PD debug menu , pressing the other button and plugging in the cable goes to DFU mode!, is the PD debug menu even supported on TS80P?

Ralim, 2023-06-27:
Hello,
Due to space constraints the debug menu is NOT available on the TS80P at this point in time 😢. I'm sorry :/

He1my, 2023-06-27:
was the debug menu available before for the TS80P and then got removed due to space constraints?
if so which previous version has it?
I just need to get a list of the reported PD voltages from the power source

discip, 2023-06-27:
Unfortunately there is no such commit, because this was never available for the **`TS80P`**.

closed 2023-06-29 by discip