Ralim/IronOS #1721: TS80P running off power bank

He1my, 2023-06-27:
using the latest release V2.21 and a dev build from 2 weeks ago https://github.com/Ralim/IronOS/suites/13720001376/artifacts/758968808

the power banks used are:

- A - [Energize UE20011PQ 22.5W](http://energizerpowerpacks.com/product_page.php?p_code=P0179&l=en)
   Rated Output:
   USB-A 5V/4.5A, 9V/2A, 12V/1.5A
   USB-C PD: 5V/2A, 9V/2A, 12V/1.5A

- B -[ hoco J80A 22.5W](https://hocotech.com/product/power/portable-chargers/power-banks/power-bank-j80a-premium-22-5w-20000mah/)
     USB-A 4.5V/5A,5V/4.5A, 9V/2A, 12V/1.5A
     USB-C PD: 5V/2.4A, 9V/2.22A, 12V/1.67A

when connected via a USB A cable the TS80P negotiates 12.3V and when starting to heat up it can reach 29W 
debug menu shows PD no VBus 

however, when connected via the USB C port it can only negotiate 9V and doesn't go beyond 18W
debug menu also shows PD no VBus 

I would assume this is by design as it is detecting the power bank can only deliver 1.5A@12V?
but if that is the case why the different behavior when connected throw the USB A port on the power bank using a USB A to USB C cable?

if that is the case and it is indeed not negotiating 12V because it is not 3A, how can I remove that limitation?
I understand that this would be overloading the power source, however, that should be decided based on the "Power limit" option under advanced settings for situations where the higher power is really needed

Ralim, 2023-06-27:
How are you measuring the 29W reading? I'm highly suspect your using the guess from ironos which will not be accurate in this situation.

Generally removing the limiter is not advisable as it results in generally either the power bank shutting off or it being overloaded leading to heat and eventually it's failure.

However if you wanted to throw caution to the wind; you could manually patch the logic in the USBPD CPP file to select 12V or you could define the "has DCDC" setting in configuration.h for the TS80P which would bypass the check on current capabilities.

He1my, 2023-06-28:
The power readings are from the detailed soldering screen, which now I understand are highly inaccurate, I am assuming the TS80P doesn't have the hardware to measure current? or it doesn't actually show the actual input voltage in the detailed soldering screen

I don't have a USB PD tester, I can strip a USB cable and use two multimeters for volts and current and an oscilloscope to monitor the actual voltage and current draw, good idea?

repeated the tests this time measuring how long the tip goes from 24c to 300c, this should give a more accurate comparison of which scenario is using more power 

```
OEM adaptor                 USB-C     12.2V    30W                10sec

18W Energizer power bank    USB-A     12.2v    30W                18sec
18W Energizer power bank    USB-C     9v       18W                19sec

22W Hoco power bank         USB-A     12.2v    30W                26sec
22W Hoco power bank         USB-C     10.2v    18W                16sec

modified the firmware, added MODEL_HAS_DCDC to allow the ts80p to negotiate 12v on a 22w power bank 
22W Hoco power bank         USB-C     12.2v    18W                40sec
```
the voltage and power readings above are what is shown on the detailed soldering screen, the power values are the maximum power shown while heating, it goes down once it reaches 300c

as can be seen from the tests modifying the firmware to bypass the check on current capabilities actually gives out the worst result, it took 40 sec to reach 300c, I think the power bank is limiting the current or going into over-current protection mode, more tests measuring the actual cable voltage and current are needed to be able to tell!

Ralim, 2023-06-29:
Yep okay, so the detailed screen is an extremely loose estimate scaling off the PWM duty and an average tip resistance. It will miss most voltage droop, or tip resistance variances etc. Its really only meant as a ballpark. Its close when used on an adequate supply.

This looks about like what is expected, spec compliant adaptors should cut you off on over-current (Unless its negotiated a PPS role, in which case it current limits you). Once you start getting non-trivial droop your going to mess up control algorithm and so the time will push out.

You could setup a test with scope to check what your seeing, but highly likely its either dropping (bad control algorithm) or its current limiting you (good control algorithm). Either way the fix here is to just use a properly rated power adaptor.

discip, 2023-06-29:
@He1my
Please don't forget to close, if your question was answered. 😊
Thanks!

closed 2023-06-30 by He1my