Ralim/IronOS #1739: WS2812B RGB mod for Pinecil V2

doegox, 2023-07-09:
Hello,

Here is a little mod to add a WS2812B to a Pinecil V2.
Obviously, it works better with the [transparent shell](https://pine64.com/product/pinecil-replacement-shell-transparent/)!

Pinecil V2 has a free GPIO_12 accessible through TP10.
I'm using it to drive a WS2812B and let the color logic already present for the MHP30 do its magic:
* green when temperature is safe (< 55°C)
* pulsing red when heating
* solid red when desired temperature is reached
* orange when cooling down (appears white on the video due to white balance)

https://github.com/Ralim/IronOS/assets/60773/ebb3e113-932a-44c1-898b-29e0118a8e5f

Electrical setup:

* WS2812B requires a Vdd between 3.5 and 5.3V and Vih (high level of input signal) must be at least 0.7Vdd
* GPIO levels are 3.3V
* The 5V rail is actually max 4.6V

So we can directly power the WS2812B on the 5V rail and command it with the GPIO without need for a level shifter, or for a Zener diode to clamp Vdd.

[image]

* Vss(3) is connected to GND, e.g. on the U10 pad mentioned and illustrated [here](https://github.com/Ralim/IronOS/issues/1410#issuecomment-1296064392)
* Vdd(1) is connected to the "5V" rail, e.g. on the capacitor mentioned and illustrated [here](https://github.com/Ralim/IronOS/issues/1410#issuecomment-1296064392)
* Din(4) is connected to TP10

Connections are made with 0.1-mm enameled wire and isolated with UV glue to avoid any shortcut (my WS2812B is not attached and could move around)

Patch is in a branch of mine for now: https://github.com/doegox/IronOS/tree/pinecilv2_ws2812b and the changes are visible in [this commit](https://github.com/doegox/IronOS/commit/2061f49b32b15dee69c4759e96820494bb45ef1c)
The driver is heavily based on WS2812.h by Ralim for the MHP30, but adapted for the RISC-V. Timings are not super precise but, well, it works...

To compile with this feature: `make firmware-EN model=Pinecilv2 ws2812b_enable=1`

Have fun!

Lennervh, 2023-07-17:
nice mod but how do you solder the wires on the PCB when it is your only soldering iron?🤯

Lennervh, 2023-07-18:
got the following error while compiling
[image]

Lennervh, 2023-07-20:
I have a problem that the led flickers instead of pulsing when heating up

doegox, 2023-07-21:
Maybe my crude bitbang timings are not good enough for your WS2812B.
You may try to monitor the signal with an oscilloscope and adjust the delay loops to be closer to the specs timings.

Lennervh, 2023-07-23:
I do not have an oscilloscope, is it possible to guess the values? Any idea what values i should try?

doegox, 2023-07-23:
no :(
an oscillo is needed to measure precisely the duration of how and high pulses and then adapt code experimentally until durations correspond as close as possible to the ones expected by WS2812B.
BTW are you sure LED is a WS2812B?
Besides flickering the LED has the proper colors ? (green when plugged, red when hot, orange when cooling down)

Lennervh, 2023-07-23:
the colors are correct the LED is a WS2812B but i used TP14 instead of TP10, change the gpio number in the pins.h

doegox, 2023-07-23:
Should be... but I didn't test it. Could you test with TP10 instead?
You made a change, the first thing is to see if that change is the source of the problem or not...

t3chguy, 2024-01-18:
TP11 and TP12 aren't usable as they're on the other side of a DNP

[image]

TP9 is usable though is under the screen, it is on the other side of the R42 DNP from TP11.
I ended up going with TP9 after TP10 broke off during my hooking up an oscilloscope to it. Turns out the build system has changed significantly and passing build flags wasn't working like before so I was probing a pinecil which wasn't trying to light an LED - D'oh. Got there eventually. TP10 shall forever be missed. My old iron is not cut out for fine work.

After much pain I got the RGB working and it is great, thanks for your initial effort @doegox 

I see no flickering and it works very nicely.

t3chguy, 2025-02-12:
@fhenning2022 

[image]

edge under the screen

Ralim, 2025-02-28:
@doegox 
If you wanted to open a PR for this, I'm happy to bring this mod in as a "feature"

Dooooor, 2025-03-01:
Is there a precompiled bin or hex file? Can't seem get this to compile without errors.

doegox, 2025-03-02:
> [@doegox](https://github.com/doegox) If you wanted to open a PR for this, I'm happy to bring this mod in as a "feature". I know its been ages, if you dont have will or time, LMK and I can probably re-base this in.

Ha thanks :)
Right now I rebased the mods in https://github.com/doegox/IronOS/tree/pinecilv2_ws2812b_2025 but there is an issue with the "new" (1 year old) GUIThread code that overrides the LED_HEATING state.
So for now I disabled it in a commit just to validate that the LED effect works as previously: https://github.com/doegox/IronOS/commit/cf6b314cd34835113e9a1234ee75bca1d6f61a48
I assume the commented code works fine with the MHP30. Any idea why it interferes with the Pinecil v2?

doegox, 2025-03-02:
> Is there a precompiled bin or hex file? Can't seem get this to compile without errors.

Try the artefacts of https://github.com/doegox/IronOS/actions/runs/13619256089

Ralim, 2025-03-03:
Likely a bug in the gui refactoring tbqh, I will look into it 🙇🏼

Ralim, 2025-03-04:
@doegox Once [this PR](https://github.com/Ralim/IronOS/pull/2094) merges led behaviour should be fixed. Apologies for the issue.

Ralim, 2025-03-05:
Merged in, whenever you have time would be great; no rush from my side (I can't talk lol)

Soopahfly, 2025-05-14:
I'm having an issue getting this to work.  I can't get the Make command to run in my Windows environment.  I'm a complete noob at this stuff.  Soldering I can do though.  Anyone have a pre-compiled FW for the Pinecil v2 with a WS2812B on TP10?

Thanks!

Ralim, 2025-05-14:
@Soopahfly This was merged into the main firmware, so you can use the auto build firmware from the [actions tab.](https://github.com/Ralim/IronOS/actions/runs/14435748292) Building on Windows is not supported (only Docker on windows).

closed 2025-05-14 by Ralim

gknops, 2025-08-07:
To answer my own question, apparently none of the builds have that enabled. I followed the Docker build instructions and added the extra flag needed, and that worked. Minor issue was https://github.com/Ralim/IronOS/issues/2149

gknops, 2025-08-07:
Thanks for this, nice feature!

barrenechea, 2026-02-12:
I recently modded my Pinecil v2, and noticed that when the screen turns off the LED is also powered off, and once the soldering iron is moved (so the screen turns on again) the LED remains off until the LED "changes into another mode" like heating.

Any chance that the LED could be continuously powered on, even if the screen is turned off?

rostovnevil-maker, 2026-09-23:
Where can I find a compiled bin file with ws2812b enabled? I only have Windows, and I can’t work with Docker.