Ralim/IronOS #1741: temperature unit entry missing in menu in DE version

HacksBugsAndRockAndRoll, 2023-07-12:
**Describe the bug**
In release version 2.21 the menu entry to change the temperature unit is missing in the DE version of the firmware

**To Reproduce**
1. flash DE version 
2. Go to 'User Interface'
4. menu entry to change temperature unit is missing (should be the first entry)

**Expected behavior**
menu entry to change temperature unit is available

**Details of your device:**
 - Device: PinecilV2
 - Release: 2.21

**Additional context**
using the EN version of the same release artifact one can find the missing config menu entry as expected

discip, 2023-07-12:
This is by design.

HacksBugsAndRockAndRoll, 2023-07-12:
This was a bit confusing to me. After getting the pinecil  (with EN firmware) I immediately updated to the latest (2.21) DE  firmware and was stuck with a combination of german menu and °F temperature unit. A factory reset helped with this but before I did this I was looking for the config option and could not find it. I went to the discord and was asked to create a ticket.

If the design assumes that DE firmware always uses °C anyways it would probably help if this was enforced by the DE firmware. Personally I would prefer having the config option available for flexibility and also consistency with the other languages.

commit ea1906e4 (2023-07-28): Enforce temp unit to C if F not in build

Closes #1741
closed 2023-07-28 by Ralim