Ralim/IronOS #1755: Only negotiates 15V if 20V profile delivers same total power

haarp, 2023-07-23:
**Describe the bug**

Hi!

I'm using a Pinecil v1 on IronOS 2.21. My powerbank (Anker PowerCore III Elite 19200) offers USB-C PD with the following profiles:

- Only Pinecil plugged in: 5/9/12/15/20V 3A
- Another device also plugged in: 5/9/12/15V 3A, 20V 2.25A

I have another device "permanently" attached to it. IronOS can decide between 20V 2.25A = 45W and 15V 3A = 45W. Under these conditions, IronOS will always negotiate 15V. 

20V would make much more sense, as it's the same amount of total power but allows faster warm-up.

I believe this is a flaw/oversight in PD negotiation.

**To Reproduce**

20V should be negotiated as it offers the same total power as 15V but with much faster warm-up

**Details of your device:**
 
 - Device: Pinecil v1
 - Release: 2.21
 - Power adapter being used: Anker PowerCore III Elite 19200

Thanks a lot! :)

Ralim, 2023-07-23:
Hello,
IronOS cares about maximum current delivery not wattage.

For IronOS to select 20V your supply must be capable of 2.5A for an 8 ohm tip and 3.3A for a 6 ohm (short) tip.

This is because the iron cannot regulate the _peak_ current used when the tip is on, only the average over a few seconds. With PD supplies spec compliant ones will shut off if we draw too high of a current through them which is generally undesirable.

haarp, 2023-07-23:
Hey Ralim, thanks for the quick reply!

This makes sense. It's quite unfortunate, because 45W 20V PD supplies are common, and miss the 2.5A mark by just a little :(

Would you be willing to implement an option to allow 2.25A supplies for 8 ohm tips? It would be out-of-spec by only 11% (not enough to trigger over-current protection), and only for short peaks.

Cheers!

Ralim, 2023-07-23:
Glad to help.

In general I'm not inclined to add an option for this, I have 3 dead PD supplies from running them _just_ over their limit here now so would rather not create more. There is absolutely nothing stopping you from making the edit if you know how.
If you don't, hit me up by email and I'm happy to make a one off build for you to test but don't want to post it publicly (for above reason). 

I hope that makes some sense, sorry its not the answer you hoped for :/

haarp, 2023-07-28:
Hey,

I found where it decides on which profile to use.

https://github.com/Ralim/IronOS/blob/55d36c98f1bb1de2c0cdce9008f5c2de9e2630ef/source/Core/Drivers/USBPD.cpp#L135-L157)

I could easily modify that, but installing a Risc compiler on my machine would be more effort ^^ I'll just leave it how it is.

I still think having the option to allow it to go slightly over spec would be sensible, but I also understand your reluctance towards it. A few dead supplies tell a tale :)

Cheers!

closed 2023-07-28 by haarp