Ralim/IronOS #1764: Multi-lang builds of Pinecilv2 fail with ld/lto1 errors

ia, 2023-07-28:
**Describe the bug**
Multi-lang builds for `Pinecilv2` fail with `ld`/`lto1` errors.

**To Reproduce**
```
$ cat test.sh
#!/usr/bin/env bash

set -x
set -e
while [ 1 -eq 1 ]; do
	make clean-build
	make  -j2  model=Pinecilv2  firmware-multi_compressed_European  firmware-multi_compressed_Bulgarian+Russian+Serbian+Ukrainian  firmware-multi_Chinese+Japanese
done;
$ make docker-shell
# ./test.sh
```

**Expected behavior**
Successful multi-lang builds for `PinecilV2`.

**Details of your device:**
Build problem, not a device one.

**Additional context**

I create this issue to:
- let others know that we're aware of the problem;
- share my thoughts & collected observations at this moment;
- figure out with help from others how to fix it in the more appropriate way.

If you work on your branch in forked repo and see similar problem, please, add comment providing:
- the direct link to the exact line with error output in the log on github CI
- copy-pasted snippet of log with error output right in the comment

Here are the examples of this issue:

[lto1 error / upstream](https://github.com/Ralim/IronOS/actions/runs/5690575529/job/15424181906#step:7:2519):
```
lto1: internal compiler error: cannot read 'LTO_section_decls' from Objects/Pinecilv2/./Core/BSP/Pinecilv2/bl_mcu_sdk/drivers/bl702_driver/hal_drv/src/hal_sec_hash.o
0xde08d0 internal_error(char const*, ...)
	???:0
0x5d8040 read_cgraph_and_symbols(unsigned int, char const**)
	???:0
0x5caf11 lto_main()
	???:0
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
lto-wrapper: fatal error: riscv-none-elf-g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:844: Hexfile/Pinecilv2_multi_compressed_Bulgarian+Russian+Serbian+Ukrainian.elf] Error 1
make[1]: Leaving directory '/__w/IronOS/IronOS/source'
make: *** [Makefile:162: firmware-multi_compressed_Bulgarian+Russian+Serbian+Ukrainian] Error 2
make: *** Waiting for unfinished jobs....
Linking Hexfile/Pinecilv2_multi_compressed_European.elf
lto1: internal compiler error: cannot read 'LTO_section_decls' from Objects/Pinecilv2/./Core/BSP/Pinecilv2/bl_mcu_sdk/drivers/bl702_driver/hal_drv/src/hal_sec_hash.o
0xde08d0 internal_error(char const*, ...)
	???:0
0x5d8040 read_cgraph_and_symbols(unsigned int, char const**)
	???:0
0x5caf11 lto_main()
	???:0
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
[... 107 lines cut ...]
INFO:root:Strings for PT compressed from 3394 to 2441 bytes (ratio 0.719)
INFO:root:Strings for SK compressed from 3454 to 2648 bytes (ratio 0.767)
INFO:root:Strings for SL compressed from 3172 to 2541 bytes (ratio 0.801)
INFO:root:Strings for SV compressed from 3206 to 2516 bytes (ratio 0.785)
INFO:root:Strings for TR compressed from 3160 to 2608 bytes (ratio 0.825)
INFO:root:Strings for VI compressed from 3300 to 2344 bytes (ratio 0.71)
INFO:root:Done
Linking Hexfile/Pinecilv2_multi_compressed_European.elf
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: warning: Objects/Pinecilv2/./Core/BSP/Pinecilv2/bl_mcu_sdk/components/ble/ble_stack/sbc/enc/sbc_analysis.o has a section extending past end of file
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: error: Objects/Pinecilv2/./Core/BSP/Pinecilv2/bl_mcu_sdk/components/ble/ble_stack/sbc/enc/sbc_analysis.o: ELF section name out of range
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:846: Hexfile/Pinecilv2_multi_compressed_European.elf] Error 1
make[1]: Leaving directory '/build/ironos/source'
make: *** [Makefile:162: firmware-multi_compressed_European] Error 2
```

Binary files mentioned in the log above [can be found here](https://github.com/ia/IronOS-plus/tree/lto1-ml-pv2/BINS).

My further plan is to:
- run builds without `-j`, with `-j2`, and with `-j4` to see the behavior;
- run builds using different ways to call `make`: `make -C source multi_...` vs `make multi_...` vs `cd source && make multi_...` (I doubt that this could be the reason but just to exclude the probability that it _somehow_ related to changing build job command for respective build on github CI).

My current suspicious that it probably may be _somehow_ related to parallel building creating race condition-like situation (i.e. some binary file _is not fully generated yet_ when _some_ related dependency in a target inside `Makefile` _"thinks"_ that it's ready.

Less (but not impossible BTW) it could be a bug in the toolchain.

ia, 2023-07-28:
#### Rate of reproducing this issue locally - about 99%. But sometimes it's _slightly_ different error every next time:

```
Linking Hexfile/Pinecilv2_multi_compressed_European.elf
lto1: fatal error: bytecode stream in file 'Objects/Pinecilv2/./Core/BSP/Pinecilv2/bl_mcu_sdk/common/partition/partition.o' generated with GCC compiler older than 10.0
compilation terminated.
lto-wrapper: fatal error: riscv-none-elf-g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/riscv-none-elf/11.2.0/../../../../riscv-none-elf/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:846: Hexfile/Pinecilv2_multi_compressed_European.elf] Error 1
make[1]: Leaving directory '/build/ironos/source'
make: *** [Makefile:162: firmware-multi_compressed_European] Error 2
```

#### **And it seems my fault after all, sorry!** :no_mouth:

---

## TL; DR - _probable_ root cause (mini write-up or _today I learned_):
- if you call `make -jX target1 target2 ... targetY` and you use only one `Makefile` then `make` tool using X for parallel build _speculatively_ will try to run needed **internal dependencies**/**internal targets** in parallel but every external **target** from input will be executed one-by-one in order, not in parallel;
- **BUT** if you call `make -jX target1 target2 ... targetY` but inside main `Makefile` you pass through input targets to be executed through another one call `make` of another one `Makefile`, then **external targets** will be run in parallel as well, i.e. `make` will just call in parallel `make target1`, `make target2` up until `make targetX` (as much as `X`s for `-j`).

---

I looked through logs before/after changes in `push.yml` very carefully and noticed that by the way how the logs report building, it seems that in the scenario with `cd source && make -j2 multi_...` **OR** in the scenario with `make -C source/` (tested locally) option for parallel build `-jN` if not ignored but applied **inside** of making every input target in [_sequential_ order](https://github.com/Ralim/IronOS/actions/runs/5690527206/job/15424031707#step:7:5) (one output of `Building for Pine64 Pinecilv2` line because `make` process is the only one).

While after the changes in `push.yml` targets themselves are [run in parallel](https://github.com/Ralim/IronOS/actions/runs/5690575529/job/15424181906#step:7:8) (two outputs of `Building for Pine64 Pinecilv2` line in parallel because `make` process forked into `nproc` processes of itself and they started to compile target `firmware-multi_compressed_European` & target `firmware-multi_compressed_Bulgarian+Russian+Serbian+Ukrainian` in parallel, hence conflict of binary data in files generated/overwritten leading to compilation error).

#### Working on a fix now...

ia, 2023-07-28:
I just added `-C source/` to `test.sh` from the original report and after that I got more than a dozen successful cycles of building. PR is ready [here](https://github.com/Ralim/IronOS/pull/1765). I had zero idea about such _**nuances**_ of behavior of `make` BTW.

Ralim, 2023-07-29:
Neither did I so I never caught it 😓 
Thanks for getting a fix figured out before I woke up; very nice to wake up to a fixed issue 😁

closed 2023-07-29 by Ralim

ia, 2023-07-29:
> Thanks for getting a fix figured out before I woke up; very nice to wake up to a fixed issue

Sure, no problem! Sorry to bring this bug in the first place to the repo. :|

And I could be wrong in the _terminology_ in root cause part but the bottom line is - as far as I could understand:
- paralleling execution of multiple `sub-targets` for main `target` which is being called by one `make` external call using one `Makefile` is good & ok;
- paralleling execution of multiple `targets` which is being called by one `make` but splitting into more than one `make` calls (one for each independent target) using the same `Makefile` _may lead_ to _nondeterministic behavior_ (as they say).

Something like that as far as I did manage to figure out this in a brief only to fix the issue in the most fast & suitable way.

Ralim, 2023-07-29:
This does make sense, but also makes it hairy to debug 😓

PR #1765 (2023-07-29): push.yml: probable fix #1764

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**
Fix #1764 bug (but maybe introduces new bugs).

* **What is the current behavior?**
Multi-lang builds for PinecilV2 crash with link-related errors due to nuances of calling multiple targets in parallel using external helper Makefile.

* **What is the new behavior (if this is a feature change)?**
Hopefully fixing crashes of multi-lang builds for PinecilV2 by calling `source/Makefil [... cut]
commit 05d0ef66 (2023-07-29): push.yml: probable fix #1764 (#1765)