Ralim/IronOS #1766: Sequre S20 Support

KosVas91, 2023-07-29:
Sequre S20 with C115 tips is a great option for precision soldering. 
As I understood @Ralim is in touch Sequre to add IronOS support for S60 model. Any chances to add S20 support too?

https://sequremall.com/products/sequre-s20-nano-electric-soldering-iron-support-pd-qc-power-supply-compatible-with-c115-soldering-iron-tips-precision-electronic-mobile-phone-repair-tool-anti-static-soldering-pen?variant=42361933889724

gamelaster, 2023-07-29:
From Sequre S60 issue, it seems that Sequre don't want to help out community to port IronOS to their devices, thus I think this will be same too. So, for proper support, we will need minimally reverse engineered datasheet.

https://github.com/Ralim/IronOS/issues/1546#issuecomment-1646819889

KosVas91, 2023-07-29:
> From Sequre S60 issue, it seems that Sequre don't want to help out community to port IronOS to their devices, thus I think this will be same too. So, for proper support, we will need minimally reverse engineered datasheet.
> 
> [#1546 (comment)](https://github.com/Ralim/IronOS/issues/1546#issuecomment-1646819889)

May be I am wrong, but the comment you mentioned is specifically about providing stock firmware for S60, not about their support in porting IronOS in general.

ia, 2023-07-29:
> the comment you mentioned is specifically about providing stock firmware for S60

Not only, it seems they don't respond to any email at all.

> not about their support in porting IronOS in general.

_Usually_ (but not necessarily) getting official firmware is _one of the steps_ to port custom firmware (by reversing the original one to see how it's done). So, in a way (by association), not providing official firmware to the community means full lack of desire to support the community in its porting free & open source projects on a company's hardware. :(

KosVas91, 2023-09-26:
Sequre has published stock firmware for S20 (and also for S60) some time ago:
https://sequremall.com/blogs/sequre-product-firmware-upgrade/s20-firmware-upgrade

team-orangeBlue, 2024-07-22:
Hey guys I noticed that the S60 is supported. Too bad I need the S20, so I got it instead.

How did the S60 get support? Reveng, convincing or something else? Can we pull off the same thing for the S20?

Ralim, 2024-07-22:
>How did the S60 get support? Reveng, convincing or something else? Can we pull off the same thing for the S20?

All of the above? 🤣 
Sequre sent me a S60 unit and partially redacted schematics (which also had errors 😁).
So I used that as a base to get roughly going and then it was reverse engineering from there.
S60P used a community member contributed PD driver and new load address offsets.
S99 is being reverse engineered at the moment in #1920.
T55 was reverse engineered (I bought a unit).
Sequre has ignored my emails to ask for schematics for S60/T55/S99/S20. I don't have the time to keep hounding them or play email tennis at this point in time.

Invictaz, 2024-11-09:
> All of the above? 🤣 Sequre sent me a S60 unit and partially redacted schematics (which also had errors 😁). So I used that as a base to get roughly going and then it was reverse engineering from [...]

So they don't reply to emails at all?