Ralim/IronOS #1774: Pencil V2: Iron OS V2.21 just black screen

AbdulKus, 2023-08-03:
Today, immediately after receiving the soldering iron, I decided to flash it for version 2.21. After flashing, the soldering iron did not always turn on when the power was connected. After a while, he completely stopped turning on with any wires and power supplies, just a black screen. After i fash to 2.20 everything worked stably. I tried again to flash to version 2.21 via blisp and via pineflash, but I only get a black screen.

gamelaster, 2023-08-03:
It is probably this issue, and it will be fixed in V2.22:
https://github.com/Ralim/IronOS/issues/1661
In mean time, you can download latest dev buiild

AbdulKus, 2023-08-04:
> It is probably this issue, and it will be fixed in V2.22: #1661 In the mean time, you can download latest dev build.

Where can I find the latest dev build?

gamelaster, 2023-08-04:
https://github.com/Ralim/IronOS/releases/tag/v2.22-rc

AbdulKus, 2023-08-04:
> https://github.com/Ralim/IronOS/releases/tag/v2.22-rc

Yey it works!

closed 2023-08-04 by AbdulKus