Ralim/IronOS #1780: Pencil V1 small display error in ZH_CN

KevinTong9, 2023-08-06:
**Describe the bug**
After turning on the details of the welding screen in the user interface settings,there will be a display problem if you try to manually turn it on after the low voltage power is connected (5V). If you turn off the detailed display, it will work perfectly.

**To Reproduce**
1. Insert normal soldering iron tip
2. 5V USB access
3. User Interface Settings
4. Open welding screen details
5. Back to Home page
6. Press the (+) button (attempt to start)
7. display failure

**Expected behavior**

should display "电压过低"

**Details of your device:**
 
 - Device:Pencil V1 (SKU:Pencil-BB2)
 - Release: v2.22H 04-08-23
 - Power adapter being used:General power bank
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**

Ralim, 2024-12-07:
Hia,
Could you please test the built files at https://github.com/Ralim/IronOS/actions/runs/12209505256

closed 2024-12-09 by Ralim

KevinTong9, 2024-12-09:
> Hia, Could you please test the built files at https://github.com/Ralim/IronOS/actions/runs/12209505256

Sorry for the late.

After trying it, it shows English “_Undervoltage VIN:4.9V_” .it works fine, but unfortunately it doesn't show Chinese while the file i  flashed is ZH_CN.dfu.

Anyway, this works perfect for me now, thank you for the patch.❤

Ralim, 2024-12-09:
>After trying it, it shows English “Undervoltage VIN:4.9V” .it works fine, but unfortunately it doesn't show Chinese while the file i flashed is ZH_CN.dfu.

The Chinese translation of the translation file is not complete; this text only has its English default text. If you are willing to help with translation, this is the file https://github.com/Ralim/IronOS/blob/dev/Translations/translation_ZH_CN.json

There are quite a few entries that still have their English default contents.
A pull request would be _ideal_ for updating this.
But if you dont want to go through that, if you can download the file and then translate the entries (i.e. the values) and upload the updated file here, that would be amazing.
If you do translate them, try to keep roughly the same length of text as the english to ensure it fits where possible. :bow:

KevinTong9, 2024-12-10:
> The Chinese translation of the translation file is not complete; this text only has its English default text. If you are willing to help with translation, this is the file [...]

PR for the translation has been submitted: [#2019 ](https://github.com/Ralim/IronOS/pull/2019). Some translations might be slightly off because Pencil V1 doesn’t have the related feature, the, but it shouldn't be too far off.