Ralim/IronOS #1781: Pinecil V2 crashes with EPR charger, v2.22H

midnightlight73, 2023-08-06:
Hello!

I'm having issues with my Pinecil v2 using an EPR charger (Aftertop 140W PD3.1 PD-089PT bought on Amazon: https://www.amazon.com/Charger-Aftertop-QC22-5W-Station-MacBook/dp/B0B3MJ5PYF)
The cable I'm using is the Anker 515 240W.

The iron crashes and reboots after about 3-5 seconds, before it ever starts to heat up.
The screen shows that the voltage drops after connecting it. I never saw it showing 28v, not even at startup. The maximum the display showed me was 24v (which makes me believe the charger starts up at 28v but drops immediately since 24v is not even a supported level)
When starting up in debug mode, the soldering iron quickly flashes through a bunch of codes, normally crashing on either 1, 11 or 12.

I don't have a different 240w cable or an other 140W/EPR device I can use to rule out an issue with my charger.
The iron does work without issues on the same charger with the same cable if I use the 100W max port (C2) instead of the 140W max port (C1).

Ralim, 2023-08-07:
Hello,
Could you confirm if this issue is present on 2.21 or 2.20?

This looks like either EPR negotiation is failing or timing out.
Ideally a capture of the PD traffic would be optimal but I understand that requires new equipment.
I'll re-test all my EPR chargers here and see if I can replicate it.

(1) Can you confirm if you are waiting for the negotiation to finish before you press to start heating (I suspect so, just checking)
(2) Any idea who makes that charger? Cant seem to find it in AU on a non-amazon seller :/

midnightlight73, 2023-08-07:
I downloaded the latest v2.22 release candidate. Before that I was running it on version v2.21.5D96470 and had the same issue.
1) I never even get to the point where I could start the heating process
2) unfortunately I don't have any more information than the one I shared.

I recorded a short video of me connecting the pinecil to the power supply, first in regular mode, than whilst holding down the + button to get into debug mode. I attach the video here, maybe you can extract some information that I missed!
I also added a closeup of the charger towards the end.
[https://photos.app.goo.gl/HqQWrC6x1a6K2up29](https://photos.app.goo.gl/HqQWrC6x1a6K2up29)

midnightlight73, 2023-09-02:
@Ralim did you manage to figure out anything or is there anything that I can do to provide you with more information?

Ralim, 2023-09-04:
As its not a regression it could be awkward to figure out.

Really next step would be to try and get a capture of whats going on with the PD bus. It looks like one or the other side is missing a message to each other and failing through to a reset.

To confirm, if you turn off EPR in the menu, does the charger then work?

discip, 2023-09-23:
Closing as @midnightlight73 doesn't seem interested anymore.

closed 2023-09-23 by discip

midnightlight73, 2023-11-26:
Hi @discip  and @Ralim very sorry for the delayed answer.
I just had the chance to try a different cable and charger. The results are:
Ugreen 300W works with both cables (Ugreen 240w nylon braided cable and Anker 515 USB 4 240w cable) and achieves the 28v. The charger mentioned in the beginning (amazon/no name model "PD-089PT") works with neither of the cables. It starts up and crashes immediately after. 
I activated and deactivated the option "PD VPDO" and changed the QC voltage setting from 14 to 22 and tried all variations with those settings. Curiously, on the Ugreen 300W charger I get the 28v regardless of whether PD VPDO is activated or not and on the noname charger the iron crashes on the 140W port regardless of the settings. It does run stable on the 100W port of the same noname charger as it always did.
I will see if I can find a different device capable of 140W to test the noname charger but until then I would say the issue seems to lie with the charger and not with the iron and there are no further actions needed from your side. Thank you for the support anyways!