Ralim/IronOS #1802: Avoid automatically heating up when inserting a new tip (pinecil v1+v2, s60; seen in Fw 1.21/1.22pre)

jakobi, 2023-08-16:
**Switching tips at runtime starts heating the new tip **

**To Reproduce**
1. iron is active and heating
2. remove old tip
3. insert new tip
4. new tip is heating up

**Expected behavior**
A switched tip should stay cool until I (long)press a button. 

Despite the 1s + duration of the bad tip detection, it currently seems to be treated as a transient read error. If indeed there are that problematic tips out there, maybe add a new boolean. Or check  the "start-heating on bootup" settings (which for me is off).

**Risk**
Use error can result in burned fingers, esp. with the s60.

Note that this is less of a risk with the hakko derived 6 or 8 ohm tips: they are slow enough to rescue each of your fingers before really hurting yourself. And you do not need to use force in seating tips. Easy...

But the s60 shows the same behaviour, and on dc12v that thing is **FAST** without power-limits being set: you'd better be **VERY AFRAID** and do NOT TOUCH the tip at all at any point in time, just as with jbc stations...

**Details of your devices:**
 - Device: pinecil v1
 - Release: 2.21
 - Device: pinecil v2
 - Release: 2.21
 - Device: s60
 - Release: 2.22H 04-08-23

ia, 2023-08-16:
I may be wrong but your report looks like a duplicate of the very similar issue report Ralim#907 with [open status here](https://github.com/Ralim/IronOS/issues/907).

Could you, please, check the provided link? And in case if it's exactly what you suggest, this one could be closed to avoid duplicated reports, but further interaction can be continued there with other participants who are interested in solving this behavior as well. Thanks.

jakobi, 2023-08-16:
looks like it, I checked new tip and heat, not burn  :). Will continue in the old thread.

closed 2023-08-17 by discip