Ralim/IronOS #1807: PD negotation fatal flaw

J3DAJ, 2023-08-22:
I'm new here and i don't even know if this is right place for this issue but here i am, so:

**Describe the bug**

I just bought **Anker 737 GaNPrime 140W PD 24000 mAh** Powerbank that can deliver 140W and to my supprise this is exacly what my Pencil V2 decided to draw... negotiated voltage was 28V. Long story short, iron and tip are fine after just few seconds of exposure but this is really bad behavior and could just simply ended up in broken iron/tip. I think i don't have to explain it any further :P.

**To Reproduce**

1. Plug stock Pencil V2 with firmware 2.21 to Anker 737 GaNPrime 140W PD 24000 mAh Powerbank via USB-C to C (140W cable)
2. Click on + to start the heating process
3. Watch power draw going over the scale (to 99W and starts over from 1W)
4. Possibly disconnect your iron ASAP to avoid burning down tip or/and iron itself

**Expected behavior**

Well, to not allow any more power than tip/iron are designed to handle. At least while using what suppose to be safer option that USB-C and PD should be.

**Details of your device:**

 - Device: Pencil V2 ( Green handle 88W version)
 - Release: idk now, i will fill this up tomorrow (i left my iron at work)
 - Power adapter being used: Anker 737 GaNPrime 140W PD 24000 mAh
 - If this is an accelerometer related issue, please include its model number here: No, i don't think it is.

**Additional context**
Will add photos tomorrow

Ralim, 2023-08-22:
Hi,
As per the official PinecilV2 design this device supports 28V@4.5A for a power of around 126W.
The tip is designed to handle 5A.
This is an expected feature of the product and I worked hard during the design to ensure that USB-PD EPR is fully supported.

The wattage indicator rolling over at 99W is a bug and has since been fixed in later firmware changes.

J3DAJ, 2023-08-23:
I just tested this out again. Powerbank says that Pencil is taking about 90W so you're right. Sorry for this false positive, I didn't had mental strength to test this properly so I won't end up with broken iron :P

Ralim, 2023-08-23:
The PinecilV2 is rated up to the 130W ish that we do; so yeah nothing to be alarmed about here.

closed 2023-08-23 by Ralim