Ralim/IronOS #1809: Release 2.22 - Release Candidate

dtdionne, 2023-08-28:
Failed to install on TS101

ia, 2023-08-28:
> Failed to install on TS101

Please, try to use advises from [here](https://github.com/Ralim/IronOS/discussions/1785#discussioncomment-6661204) (it's for `TS100` but valid for `TS101` as well) & [here](https://github.com/Ralim/IronOS/discussions/1775#discussioncomment-6631046).

If it won't help, then more detailed information is needed like a step-by-step description of what you do and what the current result is.

dtdionne, 2023-08-28:
Apologies, I just dropped the TS101_EN.hex file on the drive and a few min later it reconnected and was .err instead of .rdy.  I don't know how to pull logs from this.

1. Powered over usbc holding A
2. TS101 drive showed in windows
3. Pasted TS101_EN.hex
4. A few min later it reconnected and the filename was TS101_EN.err

Well good grief!  I tried this at least a dozen times, with the exact same file.  Didnt work.  Figured I might as well record it...and it worked.

I'm sorry.
https://youtu.be/AbFp-VMgfCQ?si=_WO6cpEQ9DH5OWmQ

closed 2023-08-28 by dtdionne

Ralim, 2023-08-28:
The miniware bootloader is incredibly hacky and unstable.
Any delays in the USB packets or recording of any data access can cause it to break. Some USB controllers have a very low success rate with it.