Ralim/IronOS #1810: Pinecil v2 usb PD wont go to 20V

tanellaanemagi, 2023-08-30:
USB PD wont go to 20V some chargers only go to 16v and some 18V
Both chargers work with previous firmware on 20V

**To Reproduce**
Flash 2.22 firmware.

**Expected behavior**
Every 65W pd charger should go to 20V

**Details of your device:**
 - Device: Pinecil V2
 - Power adapter being used: Nedis 65W PD Travel charger (works on 16v on this firmware)

ia, 2023-08-30:
Hello. Not an expert on that but allow me to figure out a couple of things for myself:

> USB PD wont go to 20V on 2.22 firmware **some chargers only go to 16v and some 18V**
> Both chargers work with 2.21 firmware on **20V**

How do you know that? I mean, could you clarify, please, what is the exact source of this voltage data? Is it based on what you see on the OLED screen? If so, did you try, for example, go to boost mode and/or increase soldering temperature to about ~350-380 C to see if there is any increase of voltage in this case?

As far as I understand, the way how this part works in `IronOS` _is not like drain max voltage as possible all the time_ but drain voltage as much as needed per every moment in time to support current soldering temperature stable enough with minimal fluctuations, hence the iron is not getting max voltage possible all the time but it's regulated depending on the current soldering temp.

But let's see what others have to add.

Ralim, 2023-08-30:
For 2.22 rc1 we had this commit in https://github.com/Ralim/IronOS/pull/1728 which solved an issue a user was experiencing with the framework laptop chargers. This was done wile we were still investigating.

This was then reverted in https://github.com/Ralim/IronOS/pull/1794 when it was found that it was only _one_ charger affected; not the entire model series.

There has not been a new rc tagged since the revert.

tanellaanemagi, 2023-08-31:
I found what the issue was, Somehow cable affected this tried different ones from goobay all was same, i replaced cable to another manufacturer cable (baseus) and everything works fine now.  strange is that with previous firmware it worked on 20v with goobay usb C-C cable also. I´m still waiting for original pine64 cables. i think these should be fine also.

mcginty, 2023-09-07:
For what it's worth, I also experienced this on my recently-received Pinecil V2. I also have a Pinecil V1, and tried both.

Using the same cable, factory reset default IronOS settings, using Pine64's [PinePower Desktop](https://www.pine64.org/pinepowerdesktop/) to confirm the voltages in PD mode:

| | Pinecil V1  | Pinecil V2 |
| ---- | ----------- | ------------- |
| v2.21 | :heavy_check_mark: (20V, heats)   | :heavy_check_mark:  (20V, heats)  |
| v2.22-rc | :heavy_check_mark: (20V, heats)   | :x: (reports 18V, crashes/reboots upon heating)  |

Ralim, 2024-01-11:
Hia,
Just wanted to loop back to check if these issues were still present on v2.22 (final release) for you?

tanellaanemagi, 2024-01-11:
Hi I'm still using 2.21 and havent updated, will test it out soon and give feedback.