Ralim/IronOS #1813: Add support for GDCPH 65W USB PD hot plate

Blaster1920, 2023-09-08:
I just got my hands on a " MHP30 PD65W Mini hot plate " from [aliexpress](https://www.aliexpress.com/item/1005005832868024.html). It seems like a poor's man MHP30 clone.
It cannot do heating curves, but you can setup custom PID factors.

I am able to provide a full pinout if this issue moves forward.

The MCU is a [STC8H3K32S2 ](https://www.lcsc.com/product-detail/Microcontroller-Units-MCUs-MPUs-SOCs_span-style-background-color-ff0-STC-span-Micro-STC8H3K32S2-45I-TSSOP20_C3013134.html).

This is connected to an i2c oled screen and to a USB PD chip, the [CH224K](https://www.lcsc.com/product-detail/USB-ICs_WCH-Jiangsu-Qin-Heng-CH224K_C970725.html).

The chip has a fully exposed UART port.

The NTC seems to have 110 Ohm at ambient temp, still soldered to the board. Pullup resistor reads 499 Ohm.

Images:
[PCB Top](https://ibb.co/ZTk7vM0)
[PCB Bottom](https://ibb.co/Fns5sv9)

Features requested:
Hotbed auto pid tune.
Follow temperature curve for paste.

discip, 2023-09-08:
Closing, as this is a duplicate of #1698.

closed 2023-09-08 by discip

StevenSeifried, 2023-09-08:
It seems that several variants of this 56x56mm hotplate exist, with different MCUs: https://github.com/Ralim/IronOS/discussions/1698

8051 architecture, 64 KiB flash and as far as I can figured out 8 KiB SRAM.

A small chance, that IronOS can be ported to those 56x56mm hotplates