Ralim/IronOS #1823: Pinecil v2 doesn't boot on Baseus Blade 100w 20 000 mAh

mmaciejak, 2023-10-03:
**Describe the bug**
When trying to power the device through the  Baseus Blade 100w 20 000 mAh powerbank, screen of the Pinencil stays black, sames happen when trying to boot in debug mode by holding the plus button while plugging the device in. Powerbank briefly displays 5V 0A on it's screen. Powerbank works fine with other devices, some as cable (Baseus 100W USB C). Pinencil also works fine when powered through Baseus 65W charger and boots normally when powered through PC USB C port

**Details of your device:**
 - Device: [Pinecil V2]
 - Release: [2.22.E.C0A5E244], same behaviour was present with 2.20 before upgrading

**Additional context**
Tried updating from 2.20 to 2.22 RC2, and changing PD timeout from 1-50 by 5 step and turning it off as well as turning off PD VPO

Ralim, 2023-10-06:
Hia,
Really the main thing to help debug this would be a capture of the USB-PD traffic. Without that I'm guessing to be quite honest.
If you connect the Pinecil while holding the rear button (programming) does the power supply "start up" (though of course Iron will stay turned off)?

mmaciejak, 2023-10-06:
Hey @Ralim, 
I just tried and by holding the minus button when plugging in, exactly the same thing happens. Powerbank briefly displays 5V 0A and then it seems to turn off. I don't have any hardware to capture PD traffic, unfortunately. 

I also tried powering the iron through USB A port of the same power bank with some off-brand USB A to USB C cable and it gives mixed results sometimes the iron stays at 5 V and sometimes it seems to negotiate successfully for 9V, obviously, it is not operating normally on 9V but starts heating up. Also when powered through USB A port and pencil is left idle the power bank seems to kill the charging frequently. The pencil flashes the electric plug symbol, and it sporadically reboots, I tried to see if adjusting the power pulse settings would help, and it somehow helps but even on the highest settings, when the tip gets over 100°C on idle it still reboots sporadically on idle.

resistancelion, 2025-01-25:
> **Describe the bug** When trying to power the device through the Baseus Blade 100w 20 000 mAh powerbank, screen of the Pinencil stays black, sames happen when trying to boot in debug mode by holding [...]

Have you tried with other Baseus PowerBank? I found that both Baseus Blade and Bipow are picky to PD protocols, i never could get them functional with most of PD devices, the only DIY thing that do work with these is chinese 100w trigger/charger circuit, others do not function, however i tried it with "ready-to-go" boards and not circuits with PD chip attached to FPGAs or microcontrollers.

Ralim, 2025-03-04:
>I just tried and by holding the minus button when plugging in, exactly the same thing happens. Powerbank briefly displays 5V 0A and then it seems to turn off. I don't have any hardware to capture PD traffic, unfortunately.

Sorry I'm so late.
If its doing this while holding the (-) that disables all PD functionality and the device will just act as a "dumb" device asking for 5V. Which sadly points to this being nearly impossible to solve if the power bank cant operate a basic 5V device :(