Ralim/IronOS #1825: compilation error Pinecilv2 help

goga1978, 2023-10-14:
HI.

c1: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
cc1plus: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
In file included from Core/Threads/OperatingModes/SolderingProfile.cpp:2:
Core/Threads/OperatingModes/OperatingModes.h:23:10: fatal error: pd.h: No such file or directory
   23 | #include "pd.h"
      |          ^~~~~~
compilation terminated.
make[1]: *** [Makefile:720: Objects/Pinecilv2/./Core/Threads/OperatingModes/SolderingProfile.o] ERROR 1
make[1]: РІС‹С…РѕРґ РёР· РєР°С‚Р°Р»РѕРіР° В«/home/goga/blis/IronOS/sourceВ»
make: *** [Makefile:228: firmware-EN] ERROR 2
goga@goga-Aspire-E1-572G:~/blis/IronOS$ make firmware-EN model=Pinecilv2
make  -C source/  firmware-EN
make[1]: РІС…РѕРґ РІ РєР°С‚Р°Р»РѕРі В«/home/goga/blis/IronOS/sourceВ»
Building for Pine64 Pinecilv2 
 
cc1plus: warning: ./Core/Drivers/usb-pd/include: No such file or directory [-Wmissing-include-dirs]
In file included from Core/Threads/OperatingModes/SolderingProfile.cpp:2:
Core/Threads/OperatingModes/OperatingModes.h:23:10: fatal error: pd.h: No such file or directory
   23 | #include "pd.h"
      |          ^~~~~~
compilation terminated.
make[1]: *** [Makefile:720: Objects/Pinecilv2/./Core/Threads/OperatingModes/SolderingProfile.o] ERROR 1
make[1]: РІС‹С…РѕРґ РёР· РєР°С‚Р°Р»РѕРіР° В«/home/goga/blis/IronOS/sourceВ»
make: *** [Makefile:228: firmware-EN] ERROR 2

[image]

StevenSeifried, 2023-10-14:
Did you used `git clone` with `--recursive` to clone the repository? Otherwise the submodules are not included

goga1978, 2023-10-14:
Thank you very much. Everything worked.
git clone --recursive https://github.com/Ralim/IronOS.git

discip, 2023-10-14:
@goga1978
If this is solved, please do not forget to close. 😊

Thank you!

closed 2023-10-15 by goga1978