Ralim/IronOS #1826: pinecilv2 Bootup Logo not showing

mml, 2023-10-15:
I believe I'm using the latest canonical instructions.

- I have a pinecil v2, verified authentic.
- Firmware v2.21.5D96470, 09-04-23.
- I built `blisp` myself from [HEAD](https://github.com/pine64/blisp/tree/7a85414eced71d3208d0cbffa1a9e7a6a0b5fcc1)
- I am using IronOS-Meta `img2logo.py` at [HEAD](https://github.com/Ralim/IronOS-Meta/tree/5ec3a16c3fd1b2258f70193291e84b9c2f9d4774)
- Everything appears to flash OK (using Linux), but upon reboot the normal screen appears.  No logo.

```
% python3 img2logo.py -m pinecilv2 Images/mml.bmp logo-
Converting Images/mml.bmp => logo-
% ls -l logo-* 
-rw-r--r-- 1 mml mml  1333 Oct 14 17:51 logo-mml.bmp.dfu
-rw-r--r-- 1 mml mml 11550 Oct 14 17:51 logo-mml.bmp.hex
-rw-r--r-- 1 mml mml  1333 Oct 14 17:51 logo-mml_L.bmp.dfu
-rw-r--r-- 1 mml mml 11550 Oct 14 17:51 logo-mml_L.bmp.hex
% sudo ~/src/github.com/pine64/blisp/build/tools/blisp/blisp write -c bl70x --reset logo-mml.bmp.dfu  
Sending a handshake...                                 
Handshake successful!
Getting chip info...
BootROM version 1.0.2.7, ChipID: 0000F4B3D7FDD7C4
0b / 59200 (0.00%)
4092b / 59200 (6.91%)
8184b / 59200 (13.82%)
12276b / 59200 (20.74%)
16368b / 59200 (27.65%)
20460b / 59200 (34.56%)
24552b / 59200 (41.47%)
28644b / 59200 (48.39%)
32736b / 59200 (55.30%)
36828b / 59200 (62.21%)
40920b / 59200 (69.12%)
45012b / 59200 (76.03%)
49104b / 59200 (82.95%)
53196b / 59200 (89.86%)
57288b / 59200 (96.77%)
59200b / 59200 (100.00%)
Sending a handshake...
Handshake with eflash_loader successful.
Input file identified as a .dfu file
Number DFU Targets: 1
Erasing flash for firmware, this might take a while...
Flashing the firmware 1024 bytes @ 0x000FE000...
0b / 1024 (0.00%)
1024b / 1024 (100.00%)
Checking program...
Program OK!
Resetting the chip.
Flash complete!
```

Ralim, 2023-10-15:
Bootup logo on pinecil V2 is only supported on the 2.22 beta's and onwards.

mml, 2023-10-15:
Thanks.  I'll try to upgrade the firmware tomorrow.

Ralim, 2023-10-16:
If your doing a reflash, and dont mind helping with some extra testing; if you could try the firmware from #1827 that would be amazing. (Just grab the Pinecilv2 zip from here https://github.com/Ralim/IronOS/actions/runs/6531970172)

ia, 2024-11-23:
> Thanks. I'll try to upgrade the firmware tomorrow.

Hello. Did you solve the issue? If yes, then, please, consider to close this issue. Thanks.

discip, 2024-11-23:
@ia
Since you seem to be very adept at telling machines to do machine tasks 😃, I thought I might suggest something I've seen in at least one other repo regarding issue maintenance.

What do you think about a bot closing all _**overdue**_ / _**abandoned**_ issues in a very friendly manner?

[image]

For reference, please have a look here. [orca_bot.yml](https://github.com/SoftFever/OrcaSlicer/blob/main/.github/workflows/orca_bot.yml)

ia, 2024-11-26:
> @ia Since you seem to be very adept at telling machines to do machine tasks 😃, I thought I might suggest something I've seen in at least one other repo regarding issue maintenance.
> 
> What do you think about a bot closing all _**overdue**_ / _**abandoned**_ issues in a very friendly manner?

You've just read my mind. Well, I was thinking about something like that, except that:

a) there should be a way to add exception for specific reports, since even here we have at least a couple of very old issues which are not closed and I really would like to help myself, just not having time but I have exactly the same hardware which were reports about;

b) I saw more than one bot on GitHub already actually, but some of them are too _"radical"_: they may close an issue in such manner that a user who is passing by with the same issue can't reopen it - this option is not available;

c) to configure, test, debug & enable bots like these, I think GitHub user should have **full access** to a repo configured with such bot, i.e. to be _owner_. But I will check and see if/how it could be done maybe.

closed 2025-03-04 by Ralim