Ralim/IronOS #1828: how many measurements does the ADC make between PWM pulses?

goga1978, 2023-10-15:
Hi.
How many measurements does the ADC make between PWM pulses?
Tell me where in the ADC code the measurement is made?
I can't find it.

Ralim, 2023-10-16:
> How many measurements does the ADC make between PWM pulses?

Depends on the device, its usually setup in `Setup.cpp` as ADC's are used in auto scan or injected mode where possible

> Tell me where in the ADC code the measurement is made?
For non PinecilV2 its triggered by a timer to occur during idle gap in the PWM scheduling cycle. On PinecilV2 its triggered by Timer IRQ (as cant do Timer Event -> ADC directly). Setup.cpp and IRQ.cpp files are the ones to look at.

>How is the temperature value filtered? Is the average taken?

Usually a rolling average; depends on device as to how much. Generally goal is to average only the readings in that temperature cycle to prevent lag.

goga1978, 2023-10-17:
Please, can you write Comments in the code? I'm new to c++ and it's hard for me to understand.
Do you have a debugging board ?

Ralim, 2023-10-18:
I write comments where the code is complex to follow for sure. The ADC setups are not overly hard to follow code-wise, but the ADC setups do require reference to the Reference Manuals for the chips to check how they are being setup.

>Do you have a debugging board ?

What do you mean by this? I do all my testing and dev work on the irons?

goga1978, 2023-10-18:
well, how to see what happens in steps on stm32

Ralim, 2023-10-18:
I'm not quite sure what you mean here.
For debugging I program the device, and if its a particularly messy issue I'll connect a debugger unit.

For knowing what steps are taken, I follow the chain usually.

Eg for PinecilV2 there is the schedular PWM that fires interrupts; in those interrupt handlers I call the function to start the ADC.
The adc then runs one "scan" of the inputs as defined here: https://github.com/Ralim/IronOS/blob/9c7ad43a76bba68c3b9384fa3514dab8504850b3/source/Core/BSP/Pinecilv2/Setup.cpp#L89
ADC is started here https://github.com/Ralim/IronOS/blob/9c7ad43a76bba68c3b9384fa3514dab8504850b3/source/Core/BSP/Pinecilv2/IRQ.cpp#L109

Where as on STM32 like devices (all others), where the adc is more full featured, its setup to use an event from the timer such as here: https://github.com/Ralim/IronOS/blob/9c7ad43a76bba68c3b9384fa3514dab8504850b3/source/Core/BSP/Sequre_S60/Setup.cpp#L195C54-L195C54

Where it is set to `ADC_EXTERNALTRIGINJECCONV_T2_TRGO` which means the `External Trigger Timer 2 Trigger Output`

So then if you look at Timer 2, here https://github.com/Ralim/IronOS/blob/9c7ad43a76bba68c3b9384fa3514dab8504850b3/source/Core/BSP/Sequre_S60/Setup.cpp#L372C14-L372C14
It is setup to output a trigger event on OutputCompare 1.
So that is how the ADC is started on that particular device.

closed 2023-10-21 by goga1978