Ralim/IronOS #1829: questions about the code

goga1978, 2023-10-21:
Hi.
[image]

IronOS/source/Core/BSP/Pinecilv2/IRQ.cpp

switchToFastPWM - ? 
switchToSlowPWM - ? 
it is quite possible to do without one?

holdoffTicks - delay before starting the measurement
tempMeasureTicks - time allotted for measurement 
powerPWM - ? 
totalPWM - ? 
PWM = 512 OR 256 - ?

uint16_t totalPWM = 255; // Total length of the cycle's ticks - ?

why such difficulties?

Ralim, 2023-10-21:
> it is quite possible to do without one?
Sure, but both are used to facilitate faster heat up.

Since the faster we sample the tip temperature, the less time we spend heating the tip. So the concept is that the device runs in a slow sample mode (slowPWM) during the initial heat up; then we transition to the fast sampling mode (fastPWM) once the device is up to temperature. All Devices in IronOS do this.
This is primarily a help with lower operating voltages and very minimal impact at higher voltages.

>powerPWM - ?

Maximum duration of the heating cycle in pwm ticks. This is set to 255, so tip heating values are 0-255 range

>totalPWM - ?

Summation of total pwm duration for the whole scheduling cycle.

This is `powerPWM + tempMeasureTicks + holdoffTicks` in the code

>PWM = 512 OR 256 - ?

Unsure what you mean here, the tip heat control values are 0-255 (inclusive), but the Timer itself goes for `totalPWM` ticks

>why such difficulties?

To facilitate fast heatup, and allow for different devices to have different scaling if required.

goga1978, 2023-10-21:
interrupt signal gpio24

[image]

then we wait holdoffTicks and measure the ADC ?

and where is it in the code?

Ralim, 2023-10-24:
GPIO24 is not an interrupt, please search the code for references to the GPIO24 to see how its used.
Its used for measuring the tip resistance.

All of the timing of ADC and tip control is done by the scheduler timer.

closed 2023-12-09 by Ralim