Ralim/IronOS #1834: TS101: How to switch back to the stock Firmware

explit7, 2023-11-07:
Hello guys. I would like to go back to stock Firmware, but i cant find stock TS101 on the web.

Thanks

explit7, 2023-11-07:
Can be closed. Flashing back works only from Windows, but not from Linux

closed 2023-11-07 by explit7

ia, 2023-11-07:
Hello.

> Flashing back works only from Windows, but not from Linux

I had this problem with my `TS80P` too in the very beginning. Your results may vary, but what **did help to me** is just three things which are documented in the official documentation ([here for TS80(P)](https://ralim.github.io/IronOS/Flashing/TS80%28P%29/) and [here for TS100](https://ralim.github.io/IronOS/Flashing/TS100/) in `Linux` subsections):
- (optionally) disable automount
- mount `DFU` storage of iron manually using **`msdos`** file system module, not **`vfat`** (used by default on automount usually, that's why automount should be disabled or manual `unmount`/`mount` using terminal should be tried)
- before writing firmware file to the storage, rename it to very short name with all caps (i.e., **`FW.HEX`** works for me)

After these steps, my success rate for flashing is (knocking on wood) 100%.

Unfortunately, this is a well-known problem of built-in `DFU` loader provided by _Miniware_ for their devices by default - the loader may be too _"picky"_ and _"sensitive"_ on that matter. Hope this information will help you in the future.

Ralim, 2023-11-07:
Would be good to know @explit7  what caused you to go back, incase its improvements we can plan?

explit7, 2023-11-07:
> Would be good to know @explit7 what caused you to go back, incase its improvements we can plan?
> 
> Miniware's bootloader has a bunch of bugs, and its only tested on old windows releases it seeeeems

Well on my TS100 i am very happy with IronOS. But the display resolution on the TS101 is a showstopper for me.
But i will come back!

Ralim, 2023-11-08:
Thanks for the feedback. Reworking all the GUI stuff for larger screens is the next on the list so all good; was concerned I'd missed more things 😓