Ralim/IronOS #1835: Using 18v DC input reboots PinecilV2 when starting to heat

blackketter, 2023-11-07:
Upgraded to v2.22E.C308FE8C (05-11-23) on Pinecil V2.  

Works fine with several USB-C adapters.  

Powering via a 18v DC through the barrel jack boots ok, but after pressing + it starts to heat a little bit and then reboots.

Restoring to v2.21 brings back functional heating via DC jack.

Ralim, 2023-11-07:
What is the power rating for the 18V supply? The new 2.22 firmware is more aggressive for power requirements when heating up.
The single most common cause for reboots like this is an insufficient power supply.

blackketter, 2023-11-07:
Ah, that could be it.   The PSU is rated at 1.67A @ 18V, so 30W. 

Under v2.21 the display showed that it was providing more than 40W successfully.

Is there any limit on 2.22?

Maybe there should be a device setting for maximum current/power when using DC in, since there's no way to detect the PSUs capability (like you can with PD).

blackketter, 2023-11-07:
I tried a 20v/4.5a/90w PSU under 2.22 and it worked just fine, so yep, this was 2.22 drawing more power than 2.21 and shutting down my power brick.

Thanks for the help!

closed 2023-11-07 by blackketter

blackketter, 2023-11-08:
I see now that there is an advanced setting for maximum power, sorry I missed this.

I'm still confused as to why 2.21 worked and 2.22 did not in my case, assuming that there was no maximum power limitation in place.

Ralim, 2023-11-08:
There was the same power limitation feature in there.
Its a bit more subtle than just limiting though.

We dont have a DC/DC regulator in there, instead the tip is just connected to the PSU or not.
And we are duty cycling this  to get an _average_ power level.

In 2.22 the rate at which we run that loop was doubled to help with temperature regulation.

This means that the power wave form looks like `__|---|__|---|___` where it is a series of pulses of power.

Since these pulses are now coming faster, the power supply may have had a harder time filtering over these and thus started tripping out.

Generally using a correctly rated power supply is the best option :/

blackketter, 2023-11-08:
Thanks, @ralim, that helps explain. 

I missed the 3A requirement for the DC in and I guess I got lucky that it worked at all!

Ralim, 2023-11-08:
Indeeed; it can be a function of luck + just how that exact unit was designed.
Sorry about the hassle 😓