Ralim/IronOS #1839: Animated Boot Logo. Continuously repeat animation when boot logo duration is set to infinity.

jhapke, 2023-11-16:
It would be cool if the boot logo animation continued to repeat until a button is pressed, only if the boot logo duration is set to infinity.

ia, 2023-11-17:
TL; DR: I _take it_! At least I _will try_ to experiment on this so @Ralim could have more free time for more important tasks.

Hello, @Ralim. I looked through related parts of the code very briefly so far. And I don't want to take any obligations not to let anyone down. But as far as I understand this feature _looks implementable_ unless I'm not aware of some very tricky _nuances_ as always. So I'm going to find some free time to experiment with this feature during this weekend so you could have more time for more important low-level improvements & features like that **very nasty MHP30/GD problem**. :crossed_fingers:

ia, 2023-11-19:
Status update: _it seems it works_ but I need a bit more time to do additional checks & prepare pull request. Will try be back during the day.

discip, 2023-11-20:
@ia
Are you planning to make this optional?
As I would not want this to loop.

kind regards

discip, 2023-11-20:
@ia
Are you planning on making this optional?
Since I don't want the **boot logo** to loop.

kind regards

ia, 2023-11-20:
> Since I don't want the boot logo to loop.

Don't worry.

> Are you planning on making this optional?

Yes, exactly. That was one of the reasons why it took longer than I thought.

Ralim, 2023-11-20:
An option here is we could add a flag to the logo file rather than an on-device setting. Or we could make the timeout option have two infinities; an infinite time and an infinite++ time that loops.

(just as a thought).

discip, 2023-11-20:
To save precious space I tend to vote for the first option.

But what would be the cost for that?
Would we get multiple logo files (looping once / endless)?

ia, 2023-11-20:
Builds with this feature are [ready](https://github.com/ia/IronOSf/actions/runs/6936651514#artifacts) in case if anyone would like to try.

A few notes:
- if the boot logo is animated & the setting is to infinite, then animation is being shown on repeat until a button press;
- to set "old" behavior (roll logo once & halt), use **1** option value (as in _one time_) for the animation setting in the menu;
- no changes for a static logo: both "1" and "infinity" options do the same for a static logo as before.

discip, 2023-11-23:
Done! 😊
Kudos to @ia.

closed 2023-11-23 by discip