Ralim/IronOS #1844: Quecoo T85 96w

iLLiac4, 2023-11-24:
https://www.quecoo.com/products/quecoo-electric-soldering-iron-kit-t85-96w-repair-tool-welding-solder-rework-station-heat-pencil-smart-portable-solder-iron-tips

Will this one also be compatible with the IronOS?

Ralim, 2023-11-24:
Please see other requests; need to know the MCU in use and its specifications before we can even guess.

turtle-bazon, 2023-12-18:
I think this is near impossible. There are CPU with 32Kb flash.
[image]

turtle-bazon, 2023-12-18:
This is nearly same as STM32F030K6U6

bspunker, 2024-01-23:
Hello,

Is this iron now supported?
What we need to test or try it?

Greetings

Heiko

ia, 2024-01-23:
Hello.

> Is this iron now supported?

No, sorry but there is no support for this device.

discip, 2024-01-23:
> Is this iron now supported?

Please refer to the [readme](https://github.com/Ralim/IronOS/tree/dev#), all supported devices are listed there.

[image]

discip, 2024-01-23:
> Is this iron now supported?

Please refer to the [readme](https://github.com/Ralim/IronOS/tree/dev?tab=readme-ov-file#ironos---flexible-soldering-iron-control-firmware), all supported devices are listed there.

[image]

SergijF, 2024-02-13:
> Please see other requests; need to know the MCU in use and its specifications before we can even guess.

Hello. If I order and give you such a soldering iron, can you adapt ironOS firmware for T85?

teodoro31570, 2024-02-20:
HI i heave an iron soldering T85 y plug in red says in display NO TOOL what can by???

teodoro31570, 2024-02-28:
CAN ANYONE PROVIDE AN ELECTRICAL DIAGRAM? iron soldering T85 THANKS

Ralim, 2024-02-28:
Please contact the manufacturer, we are not a support forum.
@teodoro31570

Marcovecchio, 2024-06-27:
Ralim, I have one of these irons, and I would like to help evaluating if the port is possible, and if so, help doing it. Buy I need some pointers. What info do you need initially, to see if the port is possible?

Ralim, 2024-06-27:
@Marcovecchio 
Hia,

So the standard docs list is here : https://ralim.github.io/IronOS/PortingToNewDevice/

The processor is already confirmed likely to work.
Usually my first step is to backup the existing firmware, then figure out the flash offsets to flash to, in order to work with the existing bootloader.

If your lucky, holding one of the buttons will make it show up in some form of update mode. As the chip is only a 32K flash ic most likely you will be lucky if they used the rom updater as then you get the full 32K.

Once the flash start/end addresses are known, next step is to figure out the pinout.

Your likely looking for I2C to the screen, a pwm output to the heater control, and GPIO for buttons.
Depending on how they have done the QC+PD implementation you may also have I2C to that or it could be standalone hardware with no software control.

Once a pinmap is figured out, the code change is to make a new BSP folder basically, put in the required libraries for hardware support/ boot. Then implement all the functions in the BSP*.h files in that folder somewhere.
This is often easiest if you take a similar device (Say a sequre) and modify it, as a lot of the logic will be same-same.

Marcovecchio, 2024-06-27:
Thank you very much for the great guide! I'll start the research this weekend and post the progress.

Marcovecchio, 2024-06-27:
@Ralim, I did some preliminar research already, and I have some points:

1) No reaction on plugging it to usb, with buttons pressed or not. I suppose it has no self-flash capability, and no bootloader. Since it's a 32kb only part, it makes sense. I found some test points, that I'll investigate if they are for the programming signals.

2) The docs say IronOS minimal flash memory requirement is 64kb, but this part is 32kb. Will IronOS really work on this?

3) I already found a complication, regarding OLED display: it's a 128x32 pixels, 0.87in display, which is nice. But I have two T85 irons, and I swapped the OLEDs, and it didn't work: the image got shifted half screen vertically, and wrapped around. I suspect both irons are from different batches, with different displays, and also different firmwares. Does that make sense?

Thanks!

Marcovecchio, 2024-06-28:
I managed to access the MCU with ST Link programmer, but it's protected against flash reading, Level 1, whatever it means.

Does anyone know if there's any workaround? The iron is functional, I would not like to erase it's flash and risk bricking it forever...

In the mean time, I'll study the hardware to identify where the OLED, buttons, heating are all connected.

Marcovecchio, 2024-06-29:
I did some more research, and read carefully the porting docs. I think I found at least a deal breaker for a possible port: the device is clearly not updatable externally, without opening. There's no apparent DFU, it seems the T85 was never meant to be user updatable. I will play some more with Boot0 pin, so there's hope still, but it's very small.

And according to the docs, 32kb of flash and 4kb of SRAM is not enough, this is probably another deal breaker. I'll not give up, I'll keep trying, it will serve as a STM32 learning project for me. But I really doubt it will turn into a real user flashable IronOS port.

nickon89, 2024-07-02:
Hello @Marcovecchio !
Your attempts to flash the chip are very interesting. It is unknown how to bypass the protection. My T85 says "No tool", but the heating goes to red. I think the problem is in the chip. This happened when I supplied AC 12V instead of DC 12V. I accidentally mixed up the AC power cord). Maybe you know what the problem is with "No tool"?
[image]

Marcovecchio, 2024-07-02:
No tool means there's no tip installed. You should never see this with a tip installed, and even worse: with high current flow.

If the iron starts heating immediately after power on, I think something got damaged around the power MOSFET: it may be in turn on state all the time. Or a mcu pin shorted.

But this is off topic here, we should be talking about IronOS porting to the T85 model.

turtle-bazon, 2024-07-26:
@Marcovecchio nice job! I also interested how 32Kb could be sufficient if in IronOS docs said that 64Kb minimum? But you can desolder CKS32F030K6U6 and solder CKS32F103C8T6 i think. :)

turtle-bazon, 2024-07-26:
@Ralim could you clear the question about sufficiency of 32Kb for the IronOS?

Ralim, 2024-07-28:
To make IronOS work on 32KiB will require cutting out features.
TS100 is the most "minimal" iron at the moment and it current comes in around 35KiB.

The bigger problem is the lack of ram. 4KiB of ram is a no-go as the font/oled code needs a fair bit of ram for the on the fly decompression. Likely supporting this unit would require a microcontroller swap out; which I think would drastically limit the user count / people to help maintain this unit?

discip, 2024-08-24:
As pointed out by **Marcovecchio** and **Ralim**, this won't work without replacing / modifying almost everything.
At this point, it's very obvious that this device is not going to be supported.

closed 2024-08-24 by discip

nightflasher, 2024-11-22:
I'm a bit confused about shortage of RAM and Memory "only" 32k - it's running already everything fine imho.

I must confess, the menu in this iron is already really big and has many options, like this profiles option. I might be out of the IronOS game, because the last time i used the T100 with IronOS is quite a long time ago, but i really like the BOOST function at all, but the T85 has the better handling: adding 50°C to whatever temperature is set and holds it until idle/stop boost, way better than holding a button.

I really want IronOS on every soldering -thingy- not only for the features, it's just like running doom on everything with GPIO - for the fun.

Ralim, 2024-11-22:
>I'm a bit confused about shortage of RAM and Memory "only" 32k - it's running already everything fine imho.

32K _can_ work, its just that at the moment IronOS has a lot of assets for all of the text and logic that chews up space. Also Keeping in mind that some of that 32K is likely consumed by the bootloader.

There is also a tradeoff made when writing the code between size optimising it for one device and making it general enough to work on all devices.

I never said it was _impossible_ just that its a non-trivial amount of work and out of scope for what I have time to do :(

serge24680, 2025-05-11:
> Привет[@Marcovecchio](https://github.com/Marcovecchio)! Ваши попытки прошить чип очень интересны. Неизвестно, как обойти защиту. Мой T85 пишет "Нет инструмента", но нагрев становится красным. Думаю, проблема в чипе. Это произошло, когда я подал AC 12V вместо DC 12V. Я случайно перепутал шнур питания переменного тока). Может быть, вы знаете, в чем проблема с "Нет инструмента"? [image]

Have you solved the problem?

nickon89, 2025-05-14:
I'm sorry for the off-topic
> Have you solved the problem?

Yes. Replaced the MOSFET and the 1F transistor. I couldn't find an analogue, so I installed a similar one, it worked.

[image]
[image]