Ralim/IronOS #1846: PTS-200 support

dexter323i, 2023-11-25:
Hi!

I have just received my new PTS200 soldering iron. It also supports TS100 soldering tips, it has buttons, display, accelerometer, USB PD and QC support, and an open soucre firmware. It is so similar to TS100, I totally assumed it can run IronOS. Well, it can't. 
[image]

It's current firmware: https://github.com/Eddddddddy/Songguo-PTS200
A similar question on their repo: https://github.com/Eddddddddy/Songguo-PTS200/issues/10

It uses ESP32-S2 FH4 MCU
Pinout:
[image]

Are you able to add support?
It would be very cool, as the current PTS200 FW is a bit rough around the edges.

Ralim, 2023-12-05:
Hia,
At the moment IronOS could run on this; though porting to an ESP32-S2 target will require a fair bit of work to get the esp-hal to work in the current layout of IronOS.

If someone has time to work on this; this should be a case of setting up a new BSP folder for this device and setting up `make` to be able to build for this..... Getting the idf to play well with make is the challenge here really, their buildsystem isn't designed to share :)

I probably wont have time at the moment to focus on this. If others want this emoji react to get a feel of desire please.
If _anyone_ wants to get this started it would be appreciated; I'm more than happy to get it over the line but I just dont have the time to do all the setup mess.

dexter323i, 2023-12-07:
Thank you for the answer!

Ralim, 2024-05-25:
Sorry for not getting back to this until now, life is busy on and off.
Wanted to check in to see if anyone knows of anyone who started a port?
I currently have another pair of projects I'm toying with but this would likely be the next one to look into after so want to find out what the go is. No timeline here I'm afraid as work is generally down to my motivation 😓

notverymoe, 2024-08-15:
I haven't been able to find any repos that look like existing porting work 😓 Is it particularly arduous to port to esp32? I don't really have any experience with micros

bausensi, 2024-10-14:
Don't feel confident enough to start my own port but I did add some extra features to the PTS200 firmware if someone is interested. It seems the main challenge to port IronOS to this iron is the ESP32S2 core's software support.

> I haven't been able to find any repos that look like existing porting work 😓 Is it particularly arduous to port to esp32? I don't really have any experience with micros
The official ESP-IDF SDK is powerful and easy but it really doesn't like to be embedded somewhere else. This is the main gripe of working with ESP32 devices like the PTS200. Porting would rely on someone either porting things like libopencm3 to ESP32 bypassing the official SDK or heavily reworking the IronOS device tree system with invasive hacks to shoehorn ESP-IDF compatibility

Thelm76, 2024-11-15:
> Sorry for not getting back to this until now, life is busy on and off. Wanted to check in to see if anyone knows of anyone who started a port? I currently have another pair of projects I'm toying [...]

Looks like someone was eager to do the port on this issue, I don't know if somthing has been done yet (nothing available in his github repositories) : https://github.com/Eddddddddy/Songguo-PTS200/issues/10

TheChatty, 2025-06-15:
There is another project [ESPIron-PTS200](https://github.com/vortigont/ESPIron-PTS200) based on Platform IO.

But I guess IronOS would need some [HAL](https://www.lenovo.com/us/en/glossary/hardware-abstraction-layer/) to utilize different hardware platforms.