Ralim/IronOS #1847: pts200 tip for ts101

lologolo, 2023-11-26:
i have a few questions:

1)Will it work pts200 tip (wich has basically the same form factor of ts101 tip but it has 4ohm resistance) with ts101 working on ironOS? 
https://it.aliexpress.com/item/1005005455327245.html?gatewayAdapt=glo2ita

2) where i can find the guide to flash ironOS on ts101 ?

3) (in ironOS) if a PD power supply isn't enought in term of amps to supply the current that a tip draws does the iron start or not?

ia, 2023-11-26:
Hello.

> where i can find the guide to flash ironOS on ts101 ?

Although [this one](https://ralim.github.io/IronOS/Flashing/TS100/#ts100) is for `TS100`, _it should work_ for `TS101` as well.

Ralim, 2023-11-26:
The means of flashing _all Miniware_ devices is the same (grab the correct hex file for your model, then copy that to the device while in DFU mode).

>Will it work pts200 tip (wich has basically the same form factor of ts101 tip but it has 4ohm resistance) on ts101 working with ironOS?

IronOS will reject the 4 ohm resistance by default, but if your willing to do some work we can enable detection and tuning for it.

I _can't_ tell you if the TS101 will blow up with the extra current draw or not however.

>(in ironOS) if a PD power supply isn't enough in term of amps to supply the current that a tip draws does the iron start or not?

The Iron will negotiate the highest voltage profile that (1) can supply enough amps for the tip and (2) is within the Irons capabilities.
So if you use a weak supply, it will negotiate a lower supply.

lologolo, 2023-11-27:
Many thanks for the answers!!

So if I've understand correctly for example with 12V 4Ω in theory it can deliver 3A but if the power supply is max at 12V 1,9A it should ask for lower voltage 9v and so on until the current is enough?

It would be very useful to implement 4Ω tip support since the classical 8Ω tip can deliver relatively low current with small voltages.

This could enhance the soldering experience enabling more portability with 9, 12v powerbanks and also more speed of reaching the set temp.

qzydustin, 2023-11-30:
According to the pts200 manufacturer, this 4-ohm soldering iron tip uses up to 20v, which would be 5 amps of current for a total of 100w of power.

I have a Pinecil v2. If it is possible to use this 4ohm soldering iron tip, it would be a better choice at lower voltages.

However, there are still big questions about whether the Pinecil V2 can handle 5a current and how to change the profile.

Ralim, 2023-12-05:
>However, there are still big questions about whether the Pinecil V2 can handle 5a current and how to change the profile.

So, in _theory_ it could handle the 5A.

What would be required to do would be to add an option for a 4 ohm tip profile, and then test really.

I can try and make up a test image for you when I get some time.

I've also ordered one of the tips, so in  ~3 weeks I'll be able to test here as well.

Ralim, 2023-12-05:
>So if I've understand correctly for example with 12V 4Ω in theory it can deliver 3A but if the power supply is max at 12V 1,9A it should ask for lower voltage 9v and so on until the current is enough?

Correct, we select the highest voltage that can supply suitable current.

Ralim, 2023-12-09:
Completely a pot-shot, but you could try this and see if the tip resistance shows up as 4 ohms.
(And then test carefully, at your risk as I haven't tested or validated this).

[Pinecilv2_EN.dfu.zip](https://github.com/Ralim/IronOS/files/13624075/Pinecilv2_EN.dfu.zip)
[TS101_EN.zip](https://github.com/Ralim/IronOS/files/13624078/TS101_EN.zip)

lologolo, 2023-12-09:
Thank you so much, when i'll receive the materials I will test, many thanks, your works is great 👍

closed 2024-03-21 by Ralim

yota9, 2024-03-31:
Hi @Ralim ! Just wondering is there a protection for PencilV2 when using 28v ERP and this 4ohm tip? It is 200w of power, I suspect iron won't be able to get over it :)

yota9, 2024-03-31:
>> Correct, we select the highest voltage that can supply suitable current.

@Ralim Hi, may I inquire is there "above" limit? E.g. When using this tip with 28 ERP it would consume almost 200w, I don't think it would survive :)

hwti, 2024-06-07:
> > > Correct, we select the highest voltage that can supply suitable current.
> 
> @Ralim Hi, may I inquire is there "above" limit? E.g. When using this tip with 28 ERP it would consume almost 200w, I don't think it would survive :) Also is it true even if we set power limit in settings? E.g. I can set 5w limit and it would still use 20v voltage. I have 96w apple adapter 20.5v and with this tip it would need more than 100w. Would setting power limit to 95w help or it would still roll back to 15v usage?

At least on the Pinecil V2, with a PTS200 tip it selects 20V instead of 28V (which would draw more than 5A).
I don't know what would happen with the DC input at 24V/28V : maybe it would limit the average current under 5A, but the peak would be higher so it might cause issues.

PR #1868 (2024-03-21): 4 ohm tip

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

This adds a 4 ohm tip profile to the TS101 and PinecilV2.
Please note that I have only tested this tip for like 5 minutes. I have not checked current ratings for the irons, so these are very much "Use at your own risk"

* **What is the current behavior?**
Closes #1847 

* **Other information**: