Ralim/IronOS #1848: MHP30 Lower minimum temperature than 100C?

Combinacijus, 2023-11-27:
Would it be possible to lower minimum temperature setpoint from 100C to let's say 0C?
I feel like there would be some use cases below 100C for a heat plate (maybe less for an iron). At very least it would be nice for testing at low temperatures which would not burn anyone or anything

Ralim, 2023-11-28:
Ahh, no your correct @ia The min temp is 10/50 testing here.
I only responded that ^ without checking the code.

@Combinacijus testing here you _can_ set the MHP30 down to 10C.

Combinacijus, 2023-11-28:
@Ralim Is 10C limit released in V2.22 already? I think I updated to it but but limit is still 100C either it didn't update correctly or it's not released yet
As for 10C it's totally ok no need for 0C as originally stated

Ralim, 2023-11-28:
That limit has been set for the last few releases

Combinacijus, 2023-11-28:
Reflashed it again and now 10C limit works. Something was wrong with flashing last time even tho it showed .RDY file.
Thanks, closing the issue as it was already solved

closed 2023-11-28 by Combinacijus