Ralim/IronOS #1849: TS80P screen orientation upside down

GrigoriOH, 2023-11-28:
**Describe the bug**

With TS80P the automatic screen orientation is upside down.

**To Reproduce**
1. Install version 2.22 on TS80P (maybe just specific devices as sometimes different motion sensors were installed)
2. Start device
3. Activate automatic screen orientation
4. Screen upside down (left handed orientation when in right hand and vice versa)

**Expected behavior**

Screen in readable orientation for auto rotation mode.

**Details of your device:**
 
 - Device: TS80P
 - Release: 2.22
 - Board version: 1.2
 - Accelerometer: LIS2DH12

**Additional context**

This issue has already been addressed in #1021 and was resolved in version 2.17

Today after work I was able to try out some previous versions and found the following:
| Version | Orientation|
|-----------|---------|
|2.22 | wrong way up|
|2.22RC3 | wrong way up|
|2.22RC2 | wrong way up|
|**2.22RC1** | **wrong way up**|
|**2.21** | **orientation is fine**|
|2.20 | orientation is fine|
|2.19 | orientation is fine|
|2.18 |orientation is fine|
|2.17 | orientation is fine (original Version of fix)|

If someone else stumbles over this issue, it might be a temporary fix to use v2.21. Also, I hope to be of some help finding the bug.

If you need some more information, please just let me know. :smiley:

damex, 2023-11-30:
hi, i can confirm that old ts80p has the same behavior on 2.22. screen is upside down.

ia, 2023-11-30:
Hello.

What is the motion sensitivity setting when the issue appears?
Have you tried to increase sensitivity to check if it solves the issue?
Does rotate not work even when you rotate the iron firmly but steady shaking it?

The bottom line is that I had minor issue with autorotate on `TS80P` in the past myself (checked on `2.22` right before the release) so I just disabled it. However, with `sensitivity = 9` it used to work for me more or less  _bearable_ anyway. So, please, try to:
- tweak motion sensitivity setting;
- rotate the iron more firmly (to trigger accelerometer for sure).

Sorry if you did it already and it didn't help but the report didn't mention anything about it. Just, please, report back if anything from the advises helped anyway.

GrigoriOH, 2023-11-30:
Hello.

>     * tweak motion sensitivity setting;
The initial setting is 7.
I just changed it to 9 but the error persists: Screen is upside down and when I rotate the device, the orientation flips, hence the displayed content is upside down again. Even with sensitivity set to 0, the screen rotates.

>     * rotate the iron more firmly (to trigger accelerometer for sure).
Both firm and soft rotation trigger the display rotation. I might have described the problem insufficiently. Auto rotation works / detects movements, but it always orients the display in the wrong direction.

>     Sorry if you did it already and it didn't help but the report didn't mention anything about it. Just, please, report back if anything from the advises helped anyway.

Nothing to be sorry about, I am grateful for any help/tip/trick as well as when I can assist you in finding a bug :smile:

Ralim, 2023-12-09:
So this is a _pain_. From what I recall from this last time it was the cloned LIS2DH12's that sometimes get into weird states depending on the order you probe the I2C bus for devices. For 2.22  I changed to a bit-banged I2C to work around miniware shipping yet-another-cloned stm device with I2C quirks.

The 2.22 firmware works fine for me here; though I have an early unit with a legit accel.

Just to rule out it being I2C timing first, can you try this file:
[TS80P_EN.zip](https://github.com/Ralim/IronOS/files/13624068/TS80P_EN.zip)

GrigoriOH, 2023-12-17:
Please excuse the late reply!
I just uploaded your firmware, but unfortunately the automated screen rotation is still the wrong way. I guess I²C timing might not be the issue.

sokerer, 2024-02-15:
I am having the same issue with 2.22, weird behaviours from the accelerometer, occasional upside down flips and no movement detections.
2.22 detects the acc as LIS2 clone

With 2.21 everything works perfectly , no issues at all 
2.21 detects the acc as SC7A20

Hope that this can give a clue

XENNOK, 2024-02-29:
Same issue here. +1
2.22 upside down
2.21 works fine

Ralim, 2024-03-18:
This was broken by TS101 support (#1695).
I should have a fix for this shortly for you to test 🙇🏼

Ralim, 2024-03-18:
If you could try flashing the build from [here](https://github.com/Ralim/IronOS/actions/runs/8320179021) and confirm if this is fixed or not would be mint.

GrigoriOH, 2024-03-19:
Can confirm, unfortunately still upside down on TS90P with this version.

Ralim, 2024-05-14:
Ah dangit.
Can you confirm if with the newer firmware the device is identified as the LIS or the LIS Clone?

discip, 2024-07-24:
@GrigoriOH
Could you please provide the requested data? 😊

thanks in advance

GrigoriOH, 2024-07-25:
Arg sorry! Somehow forgot to reply, please excuse the delay!

For Firmware *v2.22E.C308FE8C*, the ACC is reported as *LIS2 CLONE*

discip, 2024-07-25:
@Ralim
ping

Ralim, 2024-07-28:
Pushed updated fix, could you please test [this build](https://github.com/Ralim/IronOS/actions/runs/10129266983?pr=1894)

bolausson, 2024-07-31:
Fixed it for me :-)
No matter how I hod the TS80P, the display has the correct orientation now!

(Before I used to set it to "Right-handed" mode to work around this)

closed 2024-07-31 by Ralim

GrigoriOH, 2024-08-01:
Can confirm 👍  Screen orientation is correct and flips when orientation is changed.

Thanks a lot! :D

Kaizen86, 2024-09-12:
Thank you very much for fixing this, I can confirm the auto-rotation works correctly now :heart:

PR #1894 (2024-07-31): Mask LIS2DH12 clone support for TS101 only

* **Please check if the PR fulfills these requirements**
- [] The changes have been tested locally
- [] There are no breaking changes

* **What kind of change does this PR introduce?**

Trying to fix #1849 without breaking TS101

* **What is the current behavior?**

#1849
* **What is the new behavior (if this is a feature change)?**

* **Other information**: