Ralim/IronOS #1851: TS101 only working in QC mode with PD3.1 Powerbank

w00lfy111, 2023-12-05:
Hello alltogether, 
thank you for this wonderful software so far.

I´m trying to utilize PD3.1 power on my TS101 and stumbled upon some issues and discoveries.

Hardware: TS101, various tips (original as well as clone) with resistances within spec
IronOS: v2.22E.C308FE8C 05-11-23
Powerbank: INIU BI-B64 (140W PD3.1)

The combination works, but getting the full 100W out of the TS101 proves to be tricky and only works randomly so far.
Trying to heat up the Iron if the power supply is in PD mode (28.6V) will make it reboot. However if i randomly end up in QC mode (28.6V) it will proceed to heat up and function as intended.

Behavior can be reproduced, however it seems totally random if it end ups using QC or PD, with slightly better chances with a very low (1) PD timeout setting.

PD-Debug Menu shows: 28V 5A available

Working condition:
Debug Menu: PWR QC, Vin 28.6

Not Working, causing reboot while heating:
Debug Menu: PWR PD:No Vbus, Vin 28.6

For as far as i know there is no option to force the TS101 into requesting QC instead of PD power delivery?
Correct my if i´m wrong please. Would be a feature that probably solves this issue.

Any help will be appreciated, thank you!

Ralim, 2023-12-05:
QC cant do 28V; so what is likely happening is that the PD profile is "stuck" and so therefore PD negotation fails and we assume its QC.

To force QC you set PD timeout to `0`

w00lfy111, 2023-12-06:
Thank you for your quick reply. 

I tried setting the PD timeout to OFF (0 not selectable, lowest is 1 on my build) with no apparent effect.
The TS101 still cycles through the voltages from 5 to 28 displaying PD 28.6V in the end with the same problem occurring.

If it sporadically ends up displaying QC with 28V it functions as intended pulling peaks of 80-100W as indicated on the iron as well as on the powerbank. So it seems to utilize the full 28V even though it assumes it´s on QC mode.

Any advice on how to investigate this problem further?

Ralim, 2023-12-09:
>Any advice on how to investigate this problem further?

Absolute best step is to get a capture of the USB-PD packets being exchanged to debug what is going on.

otherwise next best is to boot up the iron holding the rear button to start the PD debug menu that should show what options your power supply is advertising, and make sure all advertised options match the label.
This will also show a state number that can be used to trace where its getting to in negotiating USB-PD

w00lfy111, 2023-12-12:
Unfortunately tracking the PD packets is out of reach for my expertise and equipment.

The PD-Debug Menu seems to show nohing unexpected either. Maybe you can spot something strange:
After negotiating it shows the following in the PD-Debug menu:
State 12 No VBus
1  5V  3.00A
2  9V  3.00A
3  12V  3.00A
4  15V  3.00A
5  20V  5.00A
6  3-21V  5.00A
8  28V  5.00A

Only thing odd is that that the "State 12 No VBus" display flickers every 5-10sec, and displays "State 2 No VBus"  for a fraction of a second (litterally 1/10th of a second) before rerutning to State 12.

Something else i discovered is that starting to heat immediately after plugging in the Iron, greatly increases my chances of it functioning "normally" while displaying the odd QC 28V. It seems that starting to preheat before PD is finished negotiating greatly improves the chance of working. 

Also note that it only shuts off and reboots while heating. Also the power pulse can trigger the reboot if set too high (i lowered it to mitigate that problem)

Any further advice?

ParaAdBellum, 2023-12-20:
I have a newly purchased TS101 (90W). I am experiencing similar symptom of iron rebooting while heating up on 140W Apple PD3.1 brick with EPR lead in 28v mode. This happens with V2.22, I haven't tried older firmware.
If I try another apple brick that only does 20v 5A, issue does not reproduce.
Issue also does NOT reproduce using stock firmware.

ParaAdBellum, 2024-01-14:
attaching a PD capture created using charger lab km003c and their windows software from here https://drive.google.com/drive/folders/1u63v7TAPZl0DohsziWwOkfRA4Vuk6J1Q

KM003s was inserted between TS101 and Apple 140w power brick.

[foo.zip](https://github.com/Ralim/IronOS/files/13929961/foo.zip)

it appears to be a sqlite file, which is kinda weird -- but it is viewable in their software.

if that helps, I can also capture one for stock firmware where issue is not observed.

let me know what I can do to help

ParaAdBellum, 2024-01-14:
I've collected a pd3.1 capture with ts101 running stock firmware where issue is not observed. I have a conjecture as to what's going one -- please observe two screenshots.

First screenshot is ts101 running latest ironOS:
[image]

Second screenshot is TS101 running stock:
[image]

Note how in ironOS, sink (i.e. ts101) never responds to "Vendor defined" message with value "AF 1D 01 A0 00 FF"

However, when running stock firmware sink responds to it with "Not supported" message with value "90 0C".

So the conjecture is that Apple 140w brick resets supply of voltage when it doesn't hear back from the sink with challenged with "AF 1D 01 A0 00 FF"..

thomasoteixeira, 2024-01-14:
I've experienced quite the same issues with my ts101 and ts80p as well. Both on Anker wall charger and a 100w power bank.
On the wall charger I'd see the iron negotiate to 20v but whenever I started heating it would reboot. In the power bank it would even negotiate 28v and then same behaviour - reboot on heating.
Then I found out where my issue was - or at least part of it: CABLE!
I tried all the freaking USB C cables I have, and I found out just some of them are able to negotiate (and run!) 20v, and just ONE will actually work on 28v.
I still have some weird behavior on the ts80p on USBC as it will enter a boot loop in the power bank, and sometimes even when it turns on it will not heat.
Any directions to test further are welcome, let me know and I'll be glad to help!

Ralim, 2024-01-14:
I'll have a look at the capture when I get some down time -- Thank you.
I'm curious as to why this isnt working given that the behaviour works on the PinecilV2 (where EPR was first enabled). When miniware launched the TS101 without EPR support I send them the link and a document about supporting EPR based on the IronOS code so they could have used the same code for all I know (They have stopped talking to me as of the present, so cant actually know).

I suspect you have a different / newer Apple charger to me as I dont see the vendor defined messages from my unit here.

I _think_ I know the cause of this though; so I should be able to get a draft patch figured out.

ParaAdBellum, 2024-01-14:
I've ordered a PinecilV2 a week ago. I'll see if issue reproduces with PinecilV2 once it arrives here (shipping from China).

ParaAdBellum, 2024-01-14:
Also, will be happy to try out a patch.. either pre-built or I can build one myself. thanks

ParaAdBellum, 2024-01-14:
>I tried all the freaking USB C cables I have, and I found out just some of them are able to negotiate (and run!) 20v, and just ONE will actually work on 28v.

That's kind of unsurprising:
- USB-C cables w/o E-marker chips can only do 3 amps. so at 20v that would be 60w.
- USB-C cables with E-marker chips but ones without EPR (extended power range) can only do 5A, so at 20v that would be 100w
- and then we have USB-C cables with EPR, that can do up to 5A, but are rated up to 48v, giving max power of 240W.

I haven't see any USB-PD power supplies that go above 28V yet..

Spiki22, 2024-01-23:
Hello,

I`ll give my input on this topic as well since I think I managed to encounter this problem too.
I have the Baseus 140W GaN Charger and I can use the TS101 without any problems if I use the 1m USB type C cable that was shipped in the Basesu package. However, as soon as I use another cable, with the same charger, then my Iron start rebooting while its attempting to reach the desired temperature. 

Both these cables reproduce this problem:
https://www.aliexpress.com/item/1005006184315674.html?spm=a2g0o.order_list.order_list_main.263.32621802p2dy0A
https://www.aliexpress.com/item/1005005925024987.html?spm=a2g0o.order_list.order_list_main.278.32621802p2dy0A

Please let me know if there are any logs / tests I can do in order to help you to further debug this problem.

ParaAdBellum, 2024-01-24:
> Hello,
> 
> I`ll give my input on this topic as well since I think I managed to encounter this problem too. I have the Baseus 140W GaN Charger and I can use the TS101 without any problems if I use the 1m USB type C cable that was shipped in the Basesus package. However, as soon as I use another cable, with the same charger, then my Iron start rebooting while its attempting to reach the desired temperature.
> 
> Both these cables reproduce this problem: https://www.aliexpress.com/item/1005006184315674.html?spm=a2g0o.order_list.order_list_main.263.32621802p2dy0A https://www.aliexpress.com/item/1005005925024987.html?spm=a2g0o.order_list.order_list_main.278.32621802p2dy0A
> 
> Please let me know if there are any logs / tests I can do in order to help you to further debug this problem.

this is consistent with my observations. issue is only reproducible with cables that support EPR, or power supplies that support EPR.. both of the cables you linked support EPR (i.e. are capable of going above 20v). and so is your power supply. I am venturing a guess that your original cable where issue doesn't reproduce is not an EPR cable..

Spiki22, 2024-01-24:
> this is consistent with my observations. issue is only reproducible with cables that support EPR, or power supplies that support EPR.. both of the cables you linked support EPR (i.e. are capable of [...]

But with the original cable (Basesu cable) which works without a problem, I can see that the iron is running at 28V. Shouldn't this cause the same error?

Spiki22, 2024-01-26:
Is this issue related? https://github.com/Ralim/IronOS/actions/runs/7151005424

ParaAdBellum, 2024-02-04:
> But with the original cable (Basesu cable) which works without a problem, I can see that the iron is running at 28V. Shouldn't this cause the same error?

Not entirely sure.

More data here. I received my Pinecilv2 and flashes latest 2.22 version on it, and am unable to reproduce the same issue I've observed on TS101 running same version. @Ralim is there anything else I can do to help root cause this?

Ralim, 2024-02-12:
[Archive.zip](https://github.com/Ralim/IronOS/files/14234687/Archive.zip)

Hia,
Could you try this firmware version on both devices when you have time; confirm if its the same behaviour or not on them please 🙏🏼

Spiki22, 2024-02-12:
Thanks Ralim. I have tried this version on TS101 and it's still crashing when attempting to reach temperature. If you need any other logs please let me know

ParaAdBellum, 2024-02-12:
@Ralim I've flashed provided firmware to both Ts101 and pinecilV2 and they both act about the same.. high rates of crashes while heating.

BobbyCounts, 2024-02-15:
I'm also experiencing some issues when using EPR using PinecilV2. I'm using the Apple 140W charger and two different EPR cables. On both cables the iron sometimes reboots several times before stabilizing and showing 28V is in use. It then sometimes crashes when first heating the iron. Sometimes it will work okay for a few minutes but then crash and reboot.