Ralim/IronOS #1853: Pinecil V2 won't switch to 20V on 5v/20v only capable powerbank

kitor, 2023-12-12:
**Describe the bug**
I have a Dell Power Companion PW7012M power bank with original USB-C cable. It is capable of running either 5V@1.5A or 20V@2.25A, nothing in between.

My Pinecil V2 is negotiating power and stuck in 5V. Entering PD debug menu shows both voltages available.

I'm running on firmware 2.22, but due to #1810 I tried downgrading to 2.21 - without luck. Thus I believe that's a different issue than linked report.
For testing I used 65W charger of my Lenovo laptop (capable of 20v@3.25A) - this switches into 20v immediately.

Ralim, 2023-12-12:
Hello,
20V@2.25A is not enough power, which is why 20V is  not being selected.
The tip on a Pinecil is resistive, so I=V/R (ohms law) is the rule here.
For a 6.2 ohm tip, you require a power supply that can do 20V/6.2Ohms = 3.2A
For a 8.0 ohm tip you require a power supply that can do 20V/8.0Ohms = 2.5A

2.25A is low enough that the Pinecil will never select the 20V profile.

kitor, 2023-12-12:
Thank you. It seems that ability to precisely set the maximum power limit lead me into a confusion - that device is able to control the power output directly and PSU just limits what is the maximum power available. Turns out I was wrong :)

closed 2023-12-12 by kitor

kitor, 2023-12-12:
Thinking about it - am I wrong?
How the power limit is achieved on barrel jack single voltage power supply?
Is there any resource I can read on internals? (I'm quite new to this kind of soldering irons)

Ralim, 2023-12-12:
Power limit is an average power limited for the PWM.

It limits the PWM duty cycle to keep average power below the user set limit. 

PWM does not limit the "peak" power, the mosfet is either on or off. So when it turns on, we are basically just shorting the tip resistance across the power rail. This means whenever the mosfet turns on, we pull the full current.

USB-C PSU's in general will trip off if that peak current is higher than their rated current (to within some margin).

kitor, 2023-12-12:
I see, I'm used to power regulation circuits in non resistive loads (notebooks motherboards, etc) and this is, as you mentioned - purely resistive load.
Thanks for explanation!

Ralim, 2023-12-12:
Yeah there is no inductor/diode to smooth out the current waveform, so its purely resistive in nature.

I hope it helps make some sense :)