Ralim/IronOS #1856: Disable Bluetooth by default to minimize attack vectors

rspeed, 2023-12-19:
## Description 

Bluetooth is enabled by default, which presents a potential attack vector due to the lack of authentication. Potential attacks range from pranks like setting a low temperature to an attacker potentially causing an unattended iron to start a fire.

## Steps to Reproduce

1. Reset all settings.
2. Pair with the soldering iron via Bluetooth.
3. Change settings without physical access.

## Expected Behavior

Bluetooth would either be disabled by default, or an authentication step would occur during pairing.

## Device Details

* Pinecil v2
* IronOS 2.22

Ralim, 2023-12-19:
We can default the blue interface off; however it then caused confusion for people when apps didn't work 😢.

rspeed, 2023-12-19:
Wouldn't that simply mean telling people to enable it?

Ralim, 2023-12-19:
Yeah it would, but if there is something we have learned, is that the _average_ user doesn't read half the time and instead prefers to just rant on review sites >_<.

But I do agree, even though the risk is super minimal; it is safer to default off.

ia, 2023-12-19:
I got this ;)

ia, 2023-12-19:
>> Wouldn't that simply mean telling people to enable it?

> the average user doesn't read half the time and instead prefers to just rant on review sites >_<.

I'm not android app developer myself but probably from the point of better user experience it would be nice if in `PineSAM` (is that how it's called, right?) there would be _the flow_ like this:
- user opens app and scans for `Pinecil`;
- if there is no any connection then show notification, something like: "_Please, make sure that BLE function is on on your Pinecil_";
- *or* showing this warning to a user right before the scanning/pairing attempt.

closed 2023-12-19 by Ralim

doubletaco, 2025-01-23:
Hello

I'm a blind user and really enjoy the integrations offered by the BLE functionality in the Pinecil. Community tools like [Pinesam](https://github.com/builder555/PineSAM) and [Pine Tool](https://github.com/lachlanbell/PineTool) allow me to set and monitor my tip temperatures, as well as change device settings that I otherwise could not perceive.

Is there a documented set of button presses to re-enable BLE after this update? e.g., starting from power on, how do I get to and enable BLE? And can this be setting persist across reboots?

I understand the need to disable the feature by default as a security measure. For those that rely on it, though, we should document how to re-enable it as it offers some fantastic accessibility options.

ia, 2025-01-24:
Hello, @doubletaco .

First of all, thank you for this feedback and, please, sorry for any inconvenience which may happened by disabling Bluetooth by default.

> Is there a documented set of button presses to re-enable BLE after this update? e.g., starting from power on, how do I get to and enable BLE?

This is very good hint about documenting it, since even @Ralim [did notice](https://github.com/Ralim/IronOS/issues/1856#issuecomment-1862037756) that a lot of people may be confused by not being able to make a Bluetooth pairing.

> And can this be setting persist across reboots?

Yes, of course! So, please, don't worry! Every time you change any setting in the menu, it will be saved on every exit from the menu. On top of that, **your custom settings** _should be persisted_ even between flash updates (unless there wasn't a really big rework on setting routines in the code). So the fact that now Bluetooth is disabled by default, doesn't mean that when at the next time you flash/update your Pinecil with the next stable build, your setting of Bluetooth will be off. If you have Bluetooth enabled now, this setting should survive not only reboots/turn offs, but the flash update to the next stable version.

> we should document how to re-enable it as it offers some fantastic accessibility options.

Could you, please, participate in the review, then? I'm not sure how to address getting into Settings menu in the step 1, since it depends in which state the iron is (soldering mode vs sleeping/hibernating/cooling off mode). But what do you think if the instruction will be sounded something like this:

To enable _Bluetooth_ back:
- go to _Settings_ menu;
- press `-/B` button four times to scroll the menu for `Advanced settings`;
- press `+/A` button to open submenu;
- press `+/A` button to toggle/enable _Bluetooth_ feature;
- press `-/B` **and hold it** for just more than 5 seconds to exit from the _Settings_ menu.