Ralim/IronOS #1859: How do I use a Bootup logo on TS101 on V 2.01

MaxiscooI, 2023-12-25:
I am trying to upload my own bootup logo but it isnt working because the logo just wont show i used a .bmp its less then 1kb and has the right resolution but if i plug it in upload it and disconnect it then plug it back in it just does not show the logo and the file is gone

ia, 2023-12-25:
Hello. Since there are not so many important details nor detailed step-by-step actions taken by you prior, I have to make "_interrogation_", sorry in advance. So...

> I am trying to upload my own bootup logo

Do you use built-in `DFU` loader from _Miniware_ or did you flash some 3rd party one?

Do you use "_copy-paste / drag'n'drop to dfu disk_" method to flash your logo? (I assume "yes" to this since you mentioned "_file is gone_" but it's always better to clarify)

Do you use Windows or Linux distribution or MacOS as operating system to upload? Or maybe something else? Linux & MacOS have specific order of actions to flash successfully. However, since you have _IronOS_ already, I take it as you're aware of that if you use one of those OSes.

> i used a .bmp its less then 1kb

What is _the exact full name_ of your logo file which you try to flash?

Do you try to _upload_ **raw** `bmp` file?

Did you make conversion to supported `hex` file format manually using [`img2logo.py` from here](https://github.com/Ralim/IronOS-Meta/tree/main/Bootup%20Logos) first?

Did you read [the official documentation about this feature here first](https://ralim.github.io/IronOS/Logo)?

> and has the right resolution

What do _you_ call _the right resolution_? The latest versions of _IronOS_ support `TS101` but not its "wide" screen yet, and since boot logo is a feature of _IronOS_, so logo still should be _96x16_ yet (as far as I'm aware myself).

> it just does not show the logo

Did you enable "_Show logo_" option in the settings? Even if there is a valid logo flashed, _IronOS_ doesn't show it automatically, so the option should be turned on manually.

> and the file is gone

Original file which you uploaded may be gone indeed, but do you see any `*.err` or `*.rdy` files around? 

---

Wait a minute...

> How do I use a Bootup logo on TS101 on V 2.01

What do you mean by **_V 2.01_**? _Boot logo_ is a feature of _IronOS_ so If you haven't flashed _IronOS_ first, then unfortunately we can't help you with getting boot logo on the original stock firmware. Plus, _IronOS_ **V 2.01 doesn't support TS101**.

So... could you, please, provide, additional information?

MaxiscooI, 2023-12-25:
> Hello. Since there are not so many important details nor detailed step-by-step actions taken by you prior, I have to make "_interrogation_", sorry in advance. So...
>
>
> Do you use built-in `DFU` [...]

Yes, so firstly i dont kno if it  is iron os but i think it is the normal os that comes with the soldering iron because i never changed anything. 
I drag and dropped the file , I used windows 
I used the instructions that were in the manual that came with the soldering iron it told me to make a monochrome bmp image , i did that and then it told me to copy the prepared image to the root directory of the virtual disk. I did exactly that and then when i unlugged it and plugged it again the file disappeared
I used the right resolution but not iron os i used the normal resolution (128x32) 
No i didnt see any *.err or *,rdy files
V2.01 is the firmware i updated from the soldering iron, ( i downloaded the firmware update from the official website from the soldering iron and yeah did how it told me to do it)

ia, 2023-12-25:
> Yes, so firstly i dont kno if it is iron os but i think it is the normal os that comes with the soldering iron because i never changed anything.

So, you use _the official firmware from Miniware_. But **here** it is not _technical support_ nor official forum of _Miniware_ for their firmware so you have a wrong place to ask, I'm sorry. I hope you understand that. If you have any issues with stock firmware (like boot logo issue), contact Miniware for help.

MaxiscooI, 2023-12-25:
> _ware from Miniware_. But **here** it is not _technical support_ nor official forum of _Miniware_ for their firmware so you have a wrong place to ask, I'm sorry. I hope you understand that. If you have any issues with

Do you have any idea how to contact them or where to ask??

ia, 2023-12-25:
> Do you have any idea how to contact them or where to ask??

Good question... let me google that for you:
- [the official page with the support email & contact form](https://e-design.com.cn/en/Content/2191111.html)
- [the official forum](https://forum.minidso.com/) (in Chinese, mostly)

Just keep in mind that _e-Desing/miniDSO/Miniware_ is one "umbrella" production so don't be confused with these company logos on the pages above, it's ok.

I hope now you can close this issue as _fixed_ or _misreported_.

closed 2023-12-26 by MaxiscooI

Jonny-T-cyber, 2024-02-01:
I follow the discussion