Ralim/IronOS #1872: Support Miniware MHP50?

SimGas, 2024-01-26:
**Is your feature request related to a problem? Please describe.**
Recently, the MHP50 Hotplate was released as a successor to the MHP30. It is available for about 110€ on Aliexpress and supports 100 or 150W heating, together with a new mode for reflow soldering. Little can be found on the web for this hotplate yet, but I actually expect it to get more popular than the MHP30 very soon (much larger solder plate (50x50mm instead of 30x30mm). 

However, This plate is more complex (e.g. multicolor display) than its predecessor and I am not sure if it could be covered within this project. But I just wanted to ask if you guys see any possibility supporting this plate anytime in future. 

A few things I've already noted: 
- It seems the MCU is connected to a SMD antenna, so maybe this thing theoretically has bluetooth (but currently not enabled in any part of the firmware) 
 -The Display seems to technically have a touch interface, but this is not connected.

The discussion is hereby opened :)

If anyone is curious what's inside, I've taken a few pictures:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]

ius, 2024-01-28:
MCU marking seems to read CH32F206WBU6 - a (likely) STM32 clone by Nanjing Qinheng Microelectronics (WCH). Closest match I could find on the vendor website is [CH32F208WBU6](https://www.wch-ic.com/products/CH32F208.html) which does have BLE.

I don't see any obvious debug headers (other than UART?) - so maybe it supports [SWD via USB D+/D-](https://github.com/Ralim/IronOS/issues/822#issuecomment-826181215) like the MHP30?

Ralim, 2024-01-29:
Honestly this looks like a better design than the MHP 30.
Makes me less inclined to finish my designs for a competitor.

Same rules would apply to this being supported as anything else really:
1) Need to get Freertos running on it
2) Need docs on how to connect a debugger
3) Need a full pinout of the mcu

orzel, 2024-01-31:
For the record, there's a thread about it at eevblog: https://www.eevblog.com/forum/reviews/miniware-mhp50/

TowerDefender007, 2024-02-16:
@SimGas I am using an Anker PowerPort Speed 5 with USB-C with PD but I also tried using my Shargeek battery bank that is capable of outputting 100W on the USB-C (20V at 5A) also I can see that the MHP 50 is using around 82W from the powerbank. But the biggest problem isnt that the temperature isnt accurate (or at least it would take 10min to read the right temperature) because it turns off at some point and says that the controller overheated.
Does your fan turn on at some point or is he also never active?

SimGas, 2024-02-16:
Just tried it with 250 degrees for 8 minutes. Afterwards the fan speed was 4400r/min and mcu temperature was 41 degrees. Maybe your fan is not connected? Does it show anything in the info menu? 
[image]

TowerDefender007, 2024-02-16:
@SimGas The info menu only says 0 r/min but the funny thing is after I let it heat up and the error occured I unplugged it and plugged it back in again the fan started for a short time and then turned off again. Do you know how I can open it up to take a look at the fan because I thing the fan is the problem here?

TowerDefender007, 2024-02-16:
[image]

SimGas, 2024-02-16:
Well, there are four screws under the rubber pads on the underside. Losen those and wiggle the plate slightly. When it comes off you should be left with what you can see in photo 7.
However, if the fan comes on eventually, I think it is not broken.

TowerDefender007, 2024-02-16:
I opened it and the fan is working but there is something odd happening after rebulding it the MHP 50 turns on the fan at the start and says there is no fan connected. Then I dissambled it again and rebuild it again and now it works again.

SimGas, 2024-02-16:
Maybe the pin for reading back the fan speed has a slack joint?

TowerDefender007, 2024-02-16:
I didnt checked that. I only checked the cable because it looked a little bit crushed.

ius, 2024-02-16:
Just adding a note to document that I've tested SWD via the USB C connector on MHP50, but it doesn't work for me.

- I used a Raspberry Pi Pico with CMSIS-DAP debugger firmware and butchered a USB A <-> C cable
- MHP50 powers on just fine on 5V from the Pico, but SWD fails (at least, DPIDR read fails). Even tried booting into DFU (by holding down one of the buttons when powering on).

I'm fairly confident about the electrical part and am familiar with OpenOCD (although not with Pico).

So either _this_ particular device doesn't do SWD via USB C (seems ubiquitous on Miniware products though) or I've messed up.

Anyone else feel like giving it a go? Or perhaps someone who has torn down the device is able to tell whether the SWD pins are in fact connected to USB D+/D- on this device?

orzel, 2024-02-18:
> Honestly this looks like a better design than the MHP 30. Makes me less inclined to finish my designs for a competitor.

Exactly my thoughts.... And i'm (or I was?) also designing something.

ius, 2024-04-02:
Does anybody know whether the MHP50 can be powered using 28V on the barrel jack input (rated 19-24V)?

Other Miniware devices do, and they have (unofficially) announced PD3.1/140W support for MHP50 (which would be 28V@5A) in upcoming firmware. Tried to tell by analysing the power supply circuitry from @SimGas' photo's, but the chip markings are unreadable unfortunately.

.. which is another thing IronOS on MHP50 would be great for - but it seems we lack traction. Other than my (failed) SWD experiment I'm not to keen on tearing it down as the rubber feet seem glued on...

MurdoMcLeod, 2024-04-22:
Hi Folks,
I'm new here so forgive me in case I sin against the Git-Gods.

I got one of these MHP50-B5 hot plates a few days ago, and tried it on Friday. I spent 2 hours trying to get it to work properly. I could not get it to go above ~80'c no matter what I tweaked in the menus, and no matter the different power supplies I used (USB-C PD and DC-5255 at 24V/10A).

It turns out (after much wrangling with customer support) that Miniware have a batch of these things with trapped (or easily trapped) fan cables. Mine was one of them. See attached pictures. Turns out that any shorting from the fan wires causes the hot plate to not heat up properly.
[image]
[image]
[image]
[image]
After I un-trapped the wires and repaired them with insulating tape, and reassembled it to make sure nothing was trapped, the unit worked properly.

Like night and day - the heating with 24V input was blisteringly fast. It melted all the SMD components off a small DCDC board in 1 minute.

Hope it helps - if you have one and it is playing up, dismantle and free/repair the fan wires.
Murdo.

PS Yes the feet are well stuck down but if you're careful you can re-attach the sticky feet afterwards. Just be gentle.

danielkurek, 2024-05-26:
I have traced some connections on the PCB. I have noticed that USB-C does have only one pair of D+ and D- connected that could be the reason why first attempt of connecting SWD by @ius failed. Also D+ and D- are not directly connected to other MCU pins apart from SWD communication, they could be through some resistors or some other IC for USB communication.  

Here is what i have traced so far:

- SWDIO is connected to USB D+ (PA13)
- SWCLK is connected to USB D- (PA14)
- LED is possibly some type of addressable RGB - one pin is connected to PB15
- BOOT0 and BOOT1 are connected to GND (through 2k resistors)
- Left button -> PC8
- Right button -> PC9
- 6-pin connector to hotplate is as follows (from left (closest to power input)):
  - 1-2 is power input to heater element
  - 3-4 is GND for heater element
  - 5 is GND
  - 6 is possibly some sense pin - connected through 1k resistor to PB12 with 1k pullup
- 4 pin connector to hotplate is as follows (from top left (closest to power input)):
  - 1is GND
  - 2 is PTC thermocouple - 2R around 22C (MCU pin ??)
  - 3 unknown
  - 4 (bottom right) is connected to PA3 (100k to GND on AL50 plate, possibly identification)

ius, 2024-05-27:
@danielkurek Thanks for chiming in!

In the meantime I did get SWD to work, dumped the firmware and performed some cursory reverse engineering to patch the arbitrary limits Miniware put in place for the reflow temperature/time limits. In the process I also identified parts of the pinout.

[image]

It's a bit rough (because I only did a first pass mapping all pins I've seen referenced in code) - but I have quite some functions labeled in IDA - identifying peripherals isn't too difficult based on the configuration settings/menus.

For my current use case the (patched) Miniware firmware works quite well, but if someone is willing to invest time in actually writing code to port IronOS (@Ralim?) I might be able to spend some more time on reversing.

Display is ST7735 based by the way. External flash is just used for configuration settings.

danielkurek, 2024-06-07:
Here is updated list of the MCU pinout. The main thing missing is controlling of the mosfets for the heating element. They are controlled by TPS51601A but unfortunately the main pins are connected through unknown IC.
 
- pins yet to be determined (pins have traces from them but the routing is on inner layers):
  - PD4, PB1, PB10, PB11, PD3, PB8
- pins that seem to be not connected (someone could check if they are referenced in code somewhere, @ius?):
  - PC4, PB9, PB5, PB3, PD2, PC12, PC11, PA15
- all other pins are mentioned in the text below

---

- PB15 - LED (probably addresable)
- BOOT0 & PB2/BOOT1 - 2k to GND
- PC0 - connected to 2233 (WQFN16 IC)
  - only GND and 1 GPIO pin connected to it
  - most probably 1-wire EEPROM (for storing PCB version number/configuration?)
  - this chip is also on PCB on the hotplate
- PA9 - unpopulated LED + resistor
- PC1 - connected to IC with markings LDFQB (SOT23-5?, transistor pair?)
  - perhaps something for power or temperature measurement?
- PC2 - connected to IC with markings C5C26 (SOT23-5?, transistor?)
  - perhaps something for power or temperature measurement?
- PC10 - debug UART TX?
- PA10,PB13 - connected to IC with markings 0101 A317A (SOT23-6?)
  - PA10 - maybe to PWM pin of TPS51601A
  - PB13 - maybe to SKIP_N TPS51601A

## Power input

- PC6 - connected to IC 72K (SOT23-6?) - something to do with power input P- channel mosfets
- PC7 - connected to IC 72K (SOT23-6?) - something to do with power input P- channel mosfets
- PC3 - ADC for barrel jack (18k to center pin)
- PA2 - also ADC for power input (not sure about its purpose)

## Buttons

- PC8 - BTN left
- PC9 - BTN right

## USB-C

- PA13/SWDIO - USB D+ (only one side of C-C cable)
- PA14/SWCLK - USB D- (only one side of C-C cable)
- D+ -> PA12
- D- -> PA11
- USB PD controller - ET7301 (compatible clone of FUSB302)
  - SDA -> PB7
  - SCL -> PB6
  - INT_N -> PB4

## Hotplate

### 6-pin connector

- pin 1 is closest to power input
- pins 5,6 - 1-wire interface for EEPROM on hotplate (most probably)
  - 1k to PB12 with 1k pullup
  - connected to chip 2233 (this chip is also on main board)
- 1,2 - VDD for heating element
- 3,4 - ground for heating element
- heater is 4.1R

### 4-pin connector

- 4pin - 1,2/3,4 (from top right - closest to power input)
- 1 -> G
- 2 - connected to 1 side of a wire on hotplate (other to GND on hotplate - 2R)
  - 1.2k to opamp INB+
- 3 -> NC ?
- 4 -> PA3 NTC temp (100k at 20C to GND on AL50 hotplate)

## Fan

- 1 black gnd (furthest from power input)
- 2 red VCC
- 3 yellow pin PA0
- 4 blue pin PA1 (closest to power input)

## Op-amp

- PB0 - OUTA of opamp (from some voltage regulator? - from chip with inductor)
- PC5 - OUTB of opamp
- INB+ - pin 2 from 4-pin (NTC thermocouple - around 100k at ~21C)
- INB- - 200R to GND and through some resistor and capacitor to OUTB

## Display

- 0.96"
- 80*160
- controller: ST7735S/GC9107
- [link to aliexpress](https://www.aliexpress.us/item/1005003512048646.html)
- pins (pin 13 on left from bottom - closer to BT antenna)
  - 1 TP0
  - 2 TP1
  - 3 SDA - PA7
  - 4 SCL - PA5
  - 5 RS - PD5
  - 6 RES - PD6
  - 7 CS - PA4
  - 8 GND
  - 9 NC
  - 10 VCC
  - 11 LEDK - PA8 (through transistor)
  - 12 LEDA
  - 13 GND

## External flash

- W25Q64JVSIQ
- SPI shared with LCD display
- 1 /CS -> PB14
- 2 DO -> PA6
- 3 /WP-> NC?
- 4 GND -> GND
- 5 DI -> PA7
- 6 CLK -> PA5
- 7 /HOLD or /RESET -> VDD
- 8 VCC -> VDD

danielkurek, 2024-07-09:
The pinout of the MCU is almost complete. The main pins were traced. Pins PC1 and PC2 are probably used for charging a supercapacitor (15F, 3.8V) and powering the MCU. During testing I found out that the MCU can be run entirely from the supercap even if no power input is connected (I was really confused that the display was still working even after disconnecting the power when the MCU was halted :smile:).

I have managed to get FreeRTOS working using examples from MCU manufacturer. The example project was converted to GCC so that it can be easily compiled. You can find the project in repository [CH32F208WB-FreeRTOS-GCC](https://github.com/danielkurek/CH32F208WB-FreeRTOS-GCC)  (**beware currently the project overrides the original bootloader**). Additionally, I have fixed a fork of OpenOCD that enables flashing firmware to the MCU ([fixed version](https://github.com/danielkurek/openocd)). I will start working on porting IronOS in the next few weeks.

StevenSeifried, 2024-08-24:
> The pinout of the MCU is almost complete. The main pins were traced. Pins PC1 and PC2 are probably used for charging a supercapacitor (15F, 3.8V) and powering the MCU. During testing I found out that the MCU can be run entirely from the supercap even if no power input is connected (I was really confused that the display was still working even after disconnecting the power when the MCU was halted 😄).
> 
> I have managed to get FreeRTOS working using examples from MCU manufacturer. The example project was converted to GCC so that it can be easily compiled. You can find the project in repository [CH32F208WB-FreeRTOS-GCC](https://github.com/danielkurek/CH32F208WB-FreeRTOS-GCC) (**beware currently the project overrides the original bootloader**). Additionally, I have fixed a fork of OpenOCD that enables flashing firmware to the MCU ([fixed version](https://github.com/danielkurek/openocd)). I will start working on porting IronOS in the next few weeks.

Hi @danielkurek 
maybe it's a bit too early for that, but I wanted to ask kindly about the porting progress

danielkurek, 2024-08-25:
I did not have much time to work on it but i started to work on controlling the heating element. (Also, I have updated the above pinout with my current findings). I have not found datasheet to the IC (with markings 0101 A317A) that is between mosfet driver and some MCU pins so I am currently in progress of guessing how to drive the MCU pins to control the heating element. After that, I think that it should not be hard to get the first alpha version of IronOS working on it (with minimal functions - display, buttons, heating).

I do not have much experience in reverse engineering hardware so the progress is quite slow with my limited time. However, if someone has more time to work on this, I would be happy to help and provide all my notes on reverse engineering/development. Otherwise I will be slowly working on it in my free time.

agilum, 2024-09-20:
Hey danielkurek,

is there hope to have ironOS with freely selectable reflow max temp available on mhp50 (similar to mhp30)?

danielkurek, 2024-09-20:
@agilum I will start with a very basic firmware with only buttons, screens and constant temperature mode working. Once that is working, I will try to match the features of the MHP30 that are in IronOS. So, yes reflow temps should be freely selectable (the hardware does not limit this behaviour).

Ralim, 2024-09-20:
@danielkurek
If you get stuck on anything IronOS side let me know and I can try and help or guestimate out things. 

Both the `ST7735` and the `CH32F208WBU6` will likely need driver work put in that I can do a rough pass on (dont have hardware to test).

I don't have hardware but I'm happy to try and help with the messier parts of IronOS itself.

OpenUAS, 2025-01-26:
@danielkurek and others, just some info people landing in this issue queue: All the current **original** non opensource firmware's 2.0, The beta FW 2.01Beta we've got from the manufacturer also did not fix not saved config value issue. In December 2024 We've notified the original manufacturer Now that final v2.01 is released maybe it is fixed, did not try, no time. But if you encounter this issue still, below is a quick workaround for the non saved but changed configuration temperature of WeldTemp. 

A workaround is to open the CONFIG.TXT, change 230 to 242 if you want 242 (deg C) of the device and change the line from:

```
 ...
 WeldTemp  = 230 #(C:150~260 F:302~500)
 ...
```

To:

```
 ...
 WeldTemp  = 242 #(C:150~260 F:302~500)
 ...

```
And save the file again

This is why having an opensource firmware for this MHP50 would be so much better, not depending on OEM bugs, and the delay or never fixing of products... Thanks at all for developing a opensource firmware 🥇

Lostdotfish, 2025-11-16:
Just wondering if there was any progress on the development on this port?  The MHP50 is currently gimped by the OEM firmware as we cannot set WELD Time (time at peak temp) below 60 seconds....

https://forum.minidso.com/forum.php?mod=viewthread&tid=5748

@ius you mentioned above that you were able to patch the OEM firmware arbitrary limits.  Any chance this included a lower WELD time?  A patched 2.01 OEM firmware would tide us over until this port is completed.