Ralim/IronOS #1873: Possible bug: false live temperature spikes in v2.22 on Pinecil

builder555, 2024-01-28:
**Describe the bug**
Since the latest update I noticed live data occasionally spike to 400C
[image]

**To Reproduce**

1. Open PineSAM, or any other software that gets live temperature over BLE
2. Turn on heating on pinecil
3. Observe

You can also use this stand-alone script to get values directly:

```bash
pip install pinecil
```

```python
>>> from pinecil import find_pinecils
>>> import asyncio
>>> 
>>> async def main():
>>>     pinecils = await find_pinecils()
>>>     pinecil = pinecils[0]
>>>     await pinecil.connect()
>>>     while True:
>>>         data = await pinecil.get_live_data()
>>>         print(data['LiveTemp'])
>>>         await asyncio.sleep(0.5)
>>>
>>> asyncio.run(main())
```

**Expected behavior**
Temperature rising more-less gradually

**Details of your device:**
 - Device: Pinecil
 - Release: 2.22E.C308FE8C
 - Power adapter being used: Ugreen 65W GaN.

**Additional context**
I found it to be easier to reproduce if pinecil was in sleep mode right before activating.

Ralim, 2024-01-29:
Hia,
So this has been talked about on here before.
It came up during some of the later work on 2.22. Where basically the hardware timer, just activates at entirely at the wrong time or skips a cycle. So it means the heater stays onto the start of the adc cycle (and corrupts the measurement).
At the moment the gui just filters these out mostly as a workaround as I gave up spending time trying to mask it from happening. 

I haven't yet had any new ideas on the cause, really hope its actually a software init bug but havent found it

builder555, 2024-01-29:
@Ralim do you happen to have a link to that conversation or place in the code where it's happening? Maybe I can tinker around, see what comes up.

Ralim, 2024-01-30:
The main discussion was in https://github.com/Ralim/IronOS/pull/1827

Sorry I'm away from my computer for a while so not able to search in depth .

I spent a while trying to figure it out and trying different masks but didn't quite get it to stop occuring.

Doing things like speeding up the cycle hides it better but it was still occuring so didn't go too far down that path.

builder555, 2024-01-30:
i'll close this since it's a duplicate of that discussion.

closed 2024-01-30 by builder555