Ralim/IronOS #1875: Should soldering mode be enabled on weak PD

firstnevyn, 2024-01-30:
If I plug my pinecil into a dumb non-qc non-pd output like my pc.. the pinecil fails to power the heater circuit because 5v .5a is too little to power the device

The pc reports under voltage at 4.98V 

When connected to my phone which does pd negotiation...

PD Debug
State 12 No VBus
1 5v 0.50A

If the source is PD capable (like my samsung phone) it completes pd negotiation. it offers too little power to be useful however it's completed PD negotiation.

zejjnt, 2024-02-08:
If you connect it to a dumb 5 V adapter of course it will not get enough power to be useful.

firstnevyn, 2024-02-08:
> If you connect it to a dumb 5 V adapter of course it will not get enough power to be useful.

Yes but that's not the question the question is.. if you connect to a PD source that only offers 2.5W should the iron even engage soldering mode and attempt to get to temp?  (the current behavior is.. a PD negotiation completed.. and even if the only offered profile was 2.5W you can enable soldering mode.  should there be a minimum wattage in a pd profile that's 'acceptable'?

zejjnt, 2024-02-14:
I think that is the easiest way to do it without cluttering the code; it is also useful for troubleshooting.
Of course you could clutter the code by setting a limit or whatever but with a product like this IMO that time and energy is better spent writing some useful code instead of writing something that essentially would just be there because the user didn't read the owners manual :)

firstnevyn, 2024-02-14:
By turning on the mosfet... the instantaneous draw is 5v/6.8ohm == 1.25A which is more than the pd advertisement was (0.25A)  so if a pd supply doesn't advertise at least 5V 1.5A you shouldn't turn on that mosfet.  just like 4.95V

stephen304, 2024-07-16:
This would be useful to have, there's already an "undervoltage" detection which is nice because you know immediately what the problem is, but when using PD 5V and pulling 0.6A (3W) you end up waiting a while before it finally says thermal runaway, so having an undercurrent error condition would make diagnosing the issue quicker and give a clearer indication of what's happening.

I think there are 2 places this could be implemented which handles different causes:
* For power supplies that advertise less than 5W, it could throw an immediate error when you enter solder mode - this sounds like it might be annoying if you use a PD supply like in the previous comment that supplies more power than negotiated and maybe you know it'll supply enough power anyway so this one should probably be toggleable or maybe default off if implemented
* For power supplies that advertise enough power but maybe was built poorly / won't supply 5W when it tries to heat up, an error similar to thermal runaway could be implemented but instead of checking for whether the iron is heating up by a certain time, it would check if the wattage goes over 5W by a certain time. So if a power supply stays at 3W for too long while it's trying to heat up, you'll get an undercurrent error and know that the power supply isn't providing enough current. I would assume this would be implemented similar to the thermal runaway logic so hopefully that doesn't require touching the PD code. One consideration would be that you would want to avoid giving a thermal runaway error during this process, so maybe the thermal runaway code could adjust its expectations depending on the supplied wattage so that it's more lenient when using lower power supplies - maybe that deserves its own ticket.

If only the second one is implemented, it will handle both cases but won't be as immediate since it will only "realize" the power supply isn't specced for >5W when it stays under 5W for too long (I would assume you would reset the "too long" timer if it reaches temp so it's only counting "continuous heating" time).