Ralim/IronOS #1878: MHP30 USB-PD Boot Failure

fapping-wolf, 2024-02-10:
**Describe the bug**
I have an MHP30 running version 2.22 that is failing to boot at anything above 5V. When plugged in the device simply flashes the boot logo briefly on loop. The device does not appear to be negotiating for the proper voltage as I checked with a usb-pd sniffer which failed to boot on the non 5V power supplies. I did test with the included power supply and cable but only got a 5V boot. 

**To Reproduce**
1. Plug device in to 100W USB-PD power supply
2. Observe screen flash / Boot to 5V

**Expected behavior**
Device boots at any range of voltage above 5V

**Details of your device:**
 - Device: MHP30
 - Release: 2.22E.C308FE8C
 - Power adapter being used: UGREEN CD328, UGREEN  CD294, Stock (QC92), Computer USB
 - If this is an accelerometer related issue, please include its model number here:

**Additional context**
The device previously worked with the stock firmware and I recall it working at least once with IronOS, but my memory is fallible.

Ralim, 2024-02-10:
Hi,
If you have a PD sniffer can you check what is being seen?
Does 2.21 work for you?

fapping-wolf, 2024-02-10:
Okay so the PD sniffer is fairly basic (probably better to just call it a mode display) so I can't collect a huge amount of data from it. It doesn't boot when I plug it in between the MHP30 and the power supply, which is what indicated to me that no negotiation was happening. Or negotiation fails very early in the process. Nothing looks amiss when I run it off the 5V supplies.

Just flashed 2.21 with no change.

fapping-wolf, 2024-03-29:
As a followup this issue continues with the stock firmware re-flashed to the device. This likely not an IronOS issue then.

closed 2024-03-29 by fapping-wolf