Ralim/IronOS #1879: Overtemperature

TowerDefender007, 2024-02-15:
**Describe the bug**
The problem is that the MHP 50 always will display "overtemperature" when it heats up. 

**To Reproduce**
1. Go to Heat
2. Hold the right Button till it shows a different screen.
3. Now wait till it heats up.
4. At some point it will tell you Overtemperature and turn off the heating.

**Expected behavior**
It should start heating up to the desired temperture and then hold it till I turn it off. 

**Details of your device:**
 
 - Device: MHP 50
 - Release: 
                SV: 1.00
                HV: 0.7
                PlateType: BS50
                Res: 4.0R
 - Power adapter being used: Anker PowerPort Speed 5

TowerDefender007, 2024-02-15:
It seems that the temperture measurment is wrong because when I measure the Hotplate temperature it tells me around 180-200°C but the Software only tells me that it should have 50°C. Maybe this is the problem?

Ralim, 2024-02-15:
MHP50 is **NOT** Supported by this firmware.
I'm highly suspect you have the Miniware/Morningtools firmware on your device. As such you should contact the manufacturer for support.

closed 2024-02-15 by Ralim