Ralim/IronOS #1882: STM32 died after flash read.

therealuzr, 2024-02-20:
I happens to buy a ts100 with broken firmware, after connecting it to pc it wasn't able to install driver, like the data lines was broken (they're not). So after some research i bought a St-link v2, because i thought that the bootloader was corrupted or something. I read the chip, saved what it showed me. and then i needed to flash the bootloader for ts100, which i not found, instead i flashed a ts80 bootloader and now stm32 from my iron is not responding in software (st-link utility). okay. fast forward to this very moment. I have a new ts100, i opened it, get the stm chip out, connected it to st-link, read the flash, it showed me 00x0 or something like that, then i tried to disable read protection, it read something (not only 00x0) and i unpluged stm to assembly my iron back, and it wont turn on, i disassembled to check the stm with stlink again and it's now not responding just as the previous chip. I do not understand what exactly i did wrong, any advices would be helpful. That was my 1st and 2nd time ever work with stm. So, try to not blame me much. thx

therealuzr, 2024-02-20:
> @therealuzr Try flashing this custom bootloader from @Ralim: https://github.com/Ralim/IronOS-dfu/releases/download/v0.3/TS100.zip using the St-link. But I don't know if you will be able to flash a _.hex_ or _.dfu_ file using the ST-link. As far as I know you need the original bootloader to do so.
> 
> If I got you right, you did not backup your bootloader right. Maybe @Ralim has an idea.

yeah thx, that could help if i had it before, i believe. but now st link can't connect to the stm chip

discip, 2024-02-20:
I recently had to deal with st-link and got the advice to bridge VSSA to NRST on the chip at boot in order to connect to it.

therealuzr, 2024-02-20:
> I recently had to deal with st-link and got the advice to bridge VSSA to NRST on the chip at boot in order to connect to it.
> 
> [#1806 (reply in thread)](https://github.com/Ralim/IronOS/discussions/1806#discussioncomment-8473532)

thx, i'll try to figure that out and try tomorrow

therealuzr, 2024-09-16:
[[](https://github.com/Ralim/IronOS/issues/1882#issuecomment-1954976017)
sorry for being out for quite a time. last update on topic: i bought a weact stlink copy. this time i'm able to read and flash files. so first thing i've tried is to flash @Ralim 's custom bootloader v0.3. seems like flashing went good, yet when i connect an iron to pc it still won't install dfu drivers. diffirent usb cables and ports on my nettop i've tried

discip, 2024-09-17:
@Ralim
Could you please help? 😊

Ralim, 2024-09-17:
After flashing the bootloader do you see anything on the screen?
You should be able to hold the button to enter the bootloader similar to the original one. This can indicate the STM32 chip itself is working. If you can see the screen information, does your computer recognise the usb connection at all (windows in device manager, Linux in `dmesg`?)

therealuzr, 2024-09-17:
> After flashing the bootloader do you see anything on the screen? You should be able to hold the button to enter the bootloader similar to the original one. This can indicate the STM32 chip itself is [...]

no, nothing on a screen  it’s shows up in windows device manager like “unknown something” (descriptor fail)

therealuzr, 2024-09-17:
And it’s make no difference if i press the button while connecting or not, even if i press two simultaneously

Ralim, 2024-09-17:
If nothing on screen that would suggest the bootloader is not running. The unknown device will show up from the onboard USB pull-up, the code doesn't need to run for that. 

Likely you have an issue preventing the bootloader from running. 

To confirm you have used the stink to perform an erase, flash and read back ? 
Can you try resetting the option bytes?

therealuzr, 2024-09-17:
> If nothing on screen that would suggest the bootloader is not running. The unknown device will show up from the onboard USB pull-up, the code doesn't need to run for that.
>
> Likely you have an issue [...]

Yes, i’ve used an stlink to erase and flash the bootloader. After erase whole chip was FFFFF, and after flashing there was some data. 

About option bytes, as i said much earlier i have no experience in stm family, so some article about option bytes would be helpful

Ralim, 2024-09-18:
What are you using to flash the device (what software)?
I can try and find how you reset the option bytes in that software.

>after flashing there was some data.
Check if your software can do a read-back to read back the software that was written

therealuzr, 2024-09-18:
> What are you using to flash the device (what software)? I can try and find how you reset the option bytes in that software.
> 
> > after flashing there was some data.
> > Check if your software can do a read-back to read back the software that was written

I’m using stlink utility

and if you ask about verification (“read-back to read-back”), then stlink utility says verification ok or something similar