Ralim/IronOS #1885: Support Sequre T55 hot plate

jbur4556, 2024-02-26:
**Is your feature request related to a problem? Please describe.**
While looking for a cheaper alternative to the MHP30/MHP50 I stumbled across the Sequre T55 hotplate. I would assume (naively) that it uses a similar microcontroller to the s60/s60p. Although it looks like a knock off of another mini plate that uses a stc51 (I’m not sure what the s60p uses)

Ralim, 2024-02-27:
Hia,
It will depend entirely on the chipset used inside.
Given it looks identical to the other super cheap hotplates, I suspect it is the same thing sadly.

jbur4556, 2024-02-28:
I ended up purchasing one anyways, Looks like the controller is a CKS CKS32F103CBT6. Any chance this will work?
[image]

Ralim, 2024-02-28:
Well that chip is a lot better for support.
Doing a port to it _should_ be possible.
Would move into the category of following [Porting Requirements](https://github.com/Ralim/IronOS/blob/dev/Documentation/PortingToNewDevice.md). Specifically Note B with a microcontroller pinout needing to be figured out.
Would also need to make sure we can safely flash new code to the device. It does look like the usb is wired up to _something_ so possible its wired up.

jbur4556, 2024-02-29:
After a small amount of emailing with customer service I was able to obtain a copy of the OEM firmware, I just had to promise not to share it unfortunately. They did tell me they will post any new firmware on their website, but since this is a new product for them, there is only the one version of the firmware currently. Flashing Seams easy enough, and I was able to re-flash the original firmware without needing to open the device for any reason.

I have reached out to my contact to see if I could get a copy of the MCU pinout, If he is not able to provide that to me, I don't think this device would be too difficult to reverse engineer the pinout. I will update here once I here back.
Thanks

jbur4556, 2024-03-01:
My contact said he would provide me the full schematics for the T55 if I promised to keep it confidential. As I am unfortunately a beginner in MCU programming, maybe it would be best if someone else reach out to obtain the schematics. Someone who is willing and able to port iron OS to the T55. My contact is “eric@sequremall.com”, who, I am told, is the “ Main person in charge of R&D”

jbur4556, 2024-03-03:
Bootloader offset address: 0X08004400

jbur4556, 2024-03-03:
The bootloader size is the same as the s60, I was able to flash the firmware for the s60 onto the t55 and back to the OEM firmware. The oled screen worked (still scaled, and upside down). I didn't want to try and heat up the plate for obvious reasons, but I will try and post some pinout information if I can figure it out.

Ralim, 2024-03-18:
I've ordered one of these from Aliexpress to try out, and ideally add support for.
ETA is like 3-4 weeks though, so 🤷🏼 gonna be a bit.

boilerbots, 2024-03-21:
Awesome, just got my T55 and came here to be surprised you are already talking about it. I will use it with the stock firmware but good to here that it might have options in the future.

voodoomm, 2024-04-24:
Looks like there's a new HW revision.  Chip has the same markings as in the picture from OP, but PCB is marked with "V1.1".  It's also got a 3.5mm TRRS jack on the side.  That's connected to the buttons, I think.
[image]

damex, 2024-04-24:
> Looks like there's a new HW revision. Chip has the same markings as in the picture from OP, but PCB is marked with "V1.1". It's also got a 3.5mm TRRS jack on the side. That's connected to the buttons, I think. ![T55_v1 1](https://private-user-images.githubusercontent.com/20617321/325080890-fd43a7d1-4336-42c5-9071-5b911a281687.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5NTQ5NjYsIm5iZiI6MTcxMzk1NDY2NiwicGF0aCI6Ii8yMDYxNzMyMS8zMjUwODA4OTAtZmQ0M2E3ZDEtNDMzNi00MmM1LTkwNzEtNWI5MTFhMjgxNjg3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDEwMzEwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNlMmRiNDRjOGZkZTBhN2QzN2RlNDE5OTQ4ODE1NmYzMjY2OTZmZmQ5MGMzMGRhZjI5ZWY4M2VjZWFjMDJiMzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZZ6ZwcNfRr6Je-pM0ax3gkyIdd2sYpELSGzKiglRm2s)

i have received t55 today and it is the same as yours. yeah, we have got a new revision

voodoomm, 2024-04-27:
Suppose I can stick my DT71 in there?
[image]

Ralim, 2024-05-25:
Well this has been sitting on my desk staring at me for weeks to port it over.
Has anyone poked at the pinout yet? If not I'll try to start this one next most likely.

Ralim, 2024-06-08:
Hello,
I've started #1928 which adds support for this.
I have not looked into the 3.5mm connection yet, as I have no idea what its use is 🤣 

It seems to work ok on my unit, keen for any feedback.

StevenSeifried, 2024-06-26:
My T55 arrived today and I directly flashed IronOS onto it.

I just had time for a quick test. LGTM so far.

StevenSeifried, 2024-07-22:
> Hello,
> I've started #1928 which adds support for this.
> I have not looked into the 3.5mm connection yet, as I have no idea what its use is 🤣 
> 
> It seems to work ok on my unit, keen for any feedback.

About the 3.5mm headphone jack, I got this answer from the Sequre customer service:

The headphone jack has a parameter adjustment function, which will be explained in the upgrade In the future.

PR #1928 (2024-07-24): Sequre T55 Support

* **Please check if the PR fulfills these requirements**
- [x] The changes have been tested locally
- [x] There are no breaking changes

* **What kind of change does this PR introduce?**

Adding support for the T55 hotplate.
Closes #1885.

* **What is the current behavior?**

* **What is the new behavior (if this is a feature change)?**

* **Other information**:

Like all Sequre devices, the unit is designed to minimum cost. So there is no inductor or enough caps to filter out noise.
We get some PWM hum when running it as fast as we can to keep current regulation smooth.