Ralim/IronOS #1887: TS101 reboots with MacBook M2 Pro PD

nuschpl, 2024-02-27:
**Describe the bug**
It cycles through power state 3, then 6 and reboots.

**To Reproduce**
1. Connect to MacBook M2 Pro PD
2. Observe decvice negotiating then rebooting.
3. Connect with KeyB pressed, get PD Debug state 3, PD Debug state 6. 

**Expected behavior**
Whatever power is provided via USB PD properly negotiate it so the iron will use only supported ones.

**Details of your device:**
 - Device: TS101
 - Release: [compiled from git head on 2024 Feb 27]
 - Power adapter being used: MacBook M2 Pro Thunderbolt 4 port

nuschpl, 2024-02-27:
As a followup - I've attempted to limit artificially maximum power to 15W by modyfing `configuration.h` . Compiled the firmware and it was working - on 140W capable PD charger it was runing only at 5V max very slowly heating . This is the behaviour I would expect on Macbook but it with same firmwre it still restarts instead - seems like some PD negotiation issue similar to https://github.com/Ralim/IronOS/issues/762#issuecomment-1827620725 
I'm eager to debug it further by collecting some logs or compiling custom firmware with some changes , not sure were to start.

akubak, 2024-03-20:
@nuschpl I was able to flash mine by using an old Type-C to USB-A phone cable through a hub connected to my laptop, so that no PD info was passed along to confuse the iron.

nuschpl, 2024-03-20:
I'm aware that it should be possible but this issue is not about flashing, it's precisely about proper USB PD support

dennes544, 2024-07-20:
Can confirm this. TS101 flashed with latest artifact CI #4051: Commit [545d568], same behaviour. PD Debug state 3 then PD Debug state 6 then reboot

dennes544, 2024-07-20:
Worked with Baseus GaN2 100W adapter (2 Type-C, 2 Type-A)

neon12345, 2024-07-24:
Similar problem with an Ugreen Nexode 140W GaN. There are two usb-c ports, 100W and 140W max. With the 100W port it works and with the 140W port there are constant reboots. Also reboots with my laptop usb-c but is working with a desktop pc.

dennes544, 2024-07-27:
Also strange thing i found: If flash V2.22 from releases page, TS101 works with MBP 140W adapter, BUT if i compile firmware from tag V2.22 via github actions it stops woking again. With keyB pressed on compiled firmware it shows PD state 3 then 1 then reboot.

Qwertylex, 2024-07-29:
Same thing happens here on MacBook Pro M1

dennes544, 2024-12-22:
[V2.23 RC1](https://github.com/Ralim/IronOS/releases/tag/v2.23-rc1) and latest dev artifact still same

dennes544, 2024-12-22:
@Ralim maybe there is a way to record PD negotiation and send it to you?

EsaJaatinen, 2025-01-17:
I have this same issue with brand new TS101 and PD-088PT (Meaowxva brand from Amazon, but other brands with equal model number, specs, and looks exist). Happens also with a Macbook charger and one smaller charger as well.

The PD3.1 (USB-C1) port from the PD-088PT is able to start the TS101, but the PD debug menu shows QC power mode at 20.8V (PDO 6 = 3-21V 3A). USB-C2 and -C3 have the reboot issue.

stestagg, 2025-01-30:
I think this and #2041 are duplicates

bzogala-kio, 2025-01-31:
You can only say issue to fix is duplicate or not of another if you have correctly judged on root case of each. And we have only blind guessing so far.

stestagg, 2025-01-31:
I added that link/comment to help out anyone looking into the problem, as I’ve been trying to collect together some debugging data to help with any fixes on #2104

I’m not suggesting the issues be merged or closed, merely trying to be helpful and link together potentially related issues, this way GitHub displays a nice link in the timeline so people can easily discover potentially related issues.  And if we get a fix for one (I think we’re dependent of @Ralim here, as he seems to be the expert) then it’ll be easier for everyone to check to see if the fix applies to both. 

For what it’s worth, my TS101 exhibits the exact same PD debug state output as the OP here, and it matches the behaviour in the other issue

abjugard, 2026-08-24:
I'm seeing this same behaviour with a few Pinecil V2 as well. Boots, dies, boots, dies. Booting the Pinecil V2 with (+) held shows "State 3" -> "State 6" -> "State 12", then it reboots.

This only happens when a tip is in place, if I try without a tip installed it stays at "State 3", and normal mode boots as it should.

nuschpl, 2026-08-24:
I think we know the reason, now one needs to get URL to ghost comment above ( https://github.com/Ralim/IronOS/issues/1887#issuecomment-3392587548 ) to AI and say 'Implement'

Ralim, 2026-08-24:
I never really focused on this due to it only effecting the Apple MBP and its rapid fire PD messages. Also you can't really heat well off 5V, even the short tip will only get like 4W. When I tested it, it didn't even heat up enough to use. 

Would likely require rework of the pd library to reduce latency to keep up with the timing.

abjugard, 2026-08-24:
It behaves the same way connected to my Zendure SuperTank powerbank as well for what it's worth, and that has EPR even.