Ralim/IronOS #1899: Enable Accelerometer Data Access via Bluetooth Low Energy (BLE)

Defozo, 2024-04-03:
**Is your feature request related to a problem? Please describe.**
When using my Pinecil V2 for embedding brass inserts into 3D prints, maintaining the soldering iron perfectly perpendicular to the insertion point is essential for optimal results.

**Describe the solution you'd like**
Integrating accelerometer data accessible via Bluetooth Low Energy (BLE) would be incredibly helpful. This feature would allow me to monitor and adjust the angle of the soldering iron in real-time, ensuring precision in positioning.

**Additional context**
[image]
Source of the image: https://www.youtube.com/watch?v=KqSmCHr4fdA

Nickduino, 2024-05-25:
This is such a clever way to use the accelerometer!
I wonder if you could also deduce from the acceleration in the Z-axis when the insert is low enough

discip, 2024-07-24:
@Defozo
I think a more efficient solution would be a fixture (like these [MakerWorld](https://makerworld.com/en/search/models?keyword=insert+heat+press) or these [Printables](https://www.printables.com/search/models?q=inssert%20press&ctx=models)) that works independently of the surface you are working on, as this might be tilted in one way or the other.

discip, 2024-08-24:
@Ralim
Is the accelerometer in the **`Pinecil_2`**:

1. able to take measurements in an upright orientation **_reliably_** enought?
2. capable of making the requested task **_accurate_** enough, so that it would justify the effort to implement this?

discip, 2024-10-04:
@Ralim
ping 😅